Azure/Industrial-IoT #2205: Make TransferSubscriptionsOnReconnect configurable (via publishednodes.json)

steffenbeermann, 2024-04-05:
**Is your feature request related to a problem? Please describe.**
We recently changed to 2.9 version of the opc publisher and since than we had a problem with one of our connection to one OPC UA Server. After some investigation we found that the OPC UA Server does not support the TransferSubscriptionsRequest and crashes when the OPC Publisher sends this request. In the OPC UA Publisher 2.9 you introduces the transfer subscription on reconnect feature. 

**Describe the solution you'd like**
Be able to configure the TransferSubscriptionsOnReconnect for each endpoint in the publishednodes.json. E.G:
`
  {
    // Example for an EnpointUrl is: opc.tcp://192.168.178.26:62541/Quickstarts/ReferenceServer
    "EndpointUrl": "opc.tcp://<your_opcua_server>:<your_opcua_server_port>/<your_opcua_server_path>",
    // Allows to access the endpoint with SecurityPolicy.None when set to false (no signing and encryption applied to the OPC UA communication), default is false.
    "UseSecurity": false,
    **"UseTransferSubscriptionsOnReconnect": true,**
  
    "OpcNodes": [
      {
        // Identifies the OPC node to publish in either NodeId format (contains "ns=") or ExpandedNodeId format (contains "nsu=")
        "Id": "i=2258",
        // Specifies the sampling interval OPC Publisher requests the server to sample the node value.
        // Value expressed in milliseconds.
        "OpcSamplingInterval": 1000,
        // Specifies the publishing interval OPC Publisher requests the server to publish the node value, it will only be published if the value has changed.
        // Value expressed in milliseconds.
        "OpcPublishingInterval": 5000,
        // Specifies that there should be a heartbeat generated by OPC Publisher, this means that after the given interval the last message will be sent again with an updated SourceTimestamp value.
        // Value expressed in seconds.
        "HeartbeatInterval": 3600,
        // specifies that the first event will not generate a telemetry event, this is useful when publishing a large amount of data to prevent a event flood at startup of OPC Publisher
        "SkipFirst": false
      }
    ]
  },`

**Describe alternatives you've considered**
For now we need to deploy the OPC Publisher in Version 2.8 for this one OPC UA Server.

**Additional context**

marcschier, 2024-04-22:
New option "dtr|disabletransferonreconnect" or "DisableSubscriptionTransfer": true in published nodes configuration.

closed 2024-04-22 by marcschier

steffenbeermann, 2024-04-22:
Thank you very much for the quick implementation. We already rolled it out and it seems to have solved our problem!

marcschier, 2024-06-14:
Also in 2.9.8, just released to MCR.