Azure/Industrial-IoT #2215: Subscription .... STOPPED!

Haeavar, 2024-04-29:
**Describe the bug**
If the OpcPublisher runs several days, at some point it reports errors like  `"Subscription .... STOPPED!", "Got Keep Alive error: BadNoCommunication 'Server not responding to keep alive requests."`, `"Publish #191016, Reconnecting=False, Error: BadSessionIdInvalid"` and `"Got Keep Alive error: BadSessionIdInvalid 'Session unavailable for keep alive requests."`.

The publisher is not able to reconnect its dataSetWriters. But he is still able to read nodes from the same server like
POST /v2/read on a same node of the failed DataSetWriter.

**To Reproduce**
Configure some DataSetWriters and wait several days

**Expected behavior**
The OpcPublisher recreate the session/subscription

**Log**
```
2024-04-29T06:49:15.587Z [24-04-29 06:49:15.5877] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
2024-04-29T06:49:15.588Z       Got Keep Alive error: BadNoCommunication 'Server not responding to keep alive requests.' (04/29/2024 06:49:15:BadNoCommunication 'Server not responding to keep alive requests.'
2024-04-29T06:49:15.588Z [24-04-29 06:49:15.5877] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
2024-04-29T06:49:15.588Z       Got Keep Alive error: BadNoCommunication 'Server not responding to keep alive requests.' (04/29/2024 06:49:15:BadNoCommunication 'Server not responding to keep alive requests.'
2024-04-29T06:49:15.622Z [24-04-29 06:49:15.6225] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
2024-04-29T06:49:15.622Z       Publish #191016, Reconnecting=False, Error: BadSessionIdInvalid
2024-04-29T06:49:15.629Z [24-04-29 06:49:15.6289] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
2024-04-29T06:49:15.629Z       Got Keep Alive error: BadSessionIdInvalid 'Session unavailable for keep alive requests.' (04/29/2024 06:49:15:BadSessionIdInvalid 'Session unavailable for keep alive requests.'
2024-04-29T06:49:15.629Z [24-04-29 06:49:15.6289] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
2024-04-29T06:49:15.629Z       Got Keep Alive error: BadSessionIdInvalid 'Session unavailable for keep alive requests.' (04/29/2024 06:49:15:BadSessionIdInvalid 'Session unavailable for keep alive requests.'
2024-04-29T06:49:15.656Z [24-04-29 06:49:15.6568] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
2024-04-29T06:49:15.657Z       Subscription Z0008A0011-condition-s179:3018866238 STOPPED!
2024-04-29T06:49:15.710Z [24-04-29 06:49:15.7106] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
2024-04-29T06:49:15.710Z       Publish #188214, Reconnecting=False, Error: BadSessionIdInvalid
2024-04-29T06:49:15.744Z [24-04-29 06:49:15.7441] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
2024-04-29T06:49:15.744Z       Subscription Z0008A0011-data-default:3018866241 STOPPED!
2024-04-29T06:49:15.834Z [24-04-29 06:49:15.8343] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
2024-04-29T06:49:15.834Z       Subscription Z0008A0011-event-s179:3018866243 STOPPED!
2024-04-29T06:49:15.841Z [24-04-29 06:49:15.8412] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
2024-04-29T06:49:15.841Z       Publish #188215, Reconnecting=False, Error: BadSessionIdInvalid
2024-04-29T06:49:15.927Z [24-04-29 06:49:15.9266] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
2024-04-29T06:49:15.927Z       Subscription Z0008A0011-data-opcgateway:3018866245 STOPPED!
2024-04-29T06:49:15.952Z [24-04-29 06:49:15.9525] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
2024-04-29T06:49:15.953Z       Publish #188216, Reconnecting=False, Error: BadSessionIdInvalid
2024-04-29T06:49:16.013Z [24-04-29 06:49:16.0131] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
2024-04-29T06:49:16.013Z       Publish #188217, Reconnecting=False, Error: BadSessionIdInvalid
```

**Additional context**
OPC Publisher Version (Runtime)    : 2.9.4.2+962d6fbbc2 (.NET 8.0.3/linux-x64/OPC Stack 1.5.373.121)

marcschier, 2024-04-30:
Could you try version 2.9.6?

Haeavar, 2024-05-08:
@marcschier yes. I will let you know if the error is still there. It will take some days.

alxy, 2024-06-12:
We have experienced exactly the same behavior now with `2.9.4`.

marcschier, 2024-06-17:
Please use 2.9.8 (now) - it is the latest release.  If the issue occurs, please update this issue before 2.9.9 release, at which point I will otherwise close

alxy, 2024-06-17:
Can you tell in which version precisely this was fixed? I have just upgraded to 2.9.6 and it takes a while until this reaches our production environment (due to our internal release cycle) . So if you know something has been fixed related to his after 2.9.6, please let me know.

marcschier, 2024-06-17:
2.9.6 is fine to use, but it has these informational messages above ("Limiting number of messages"). 2.9.8 does not have those log messages anymore.

marcschier, 2024-06-18:
Moved to 2.9.10 - if not fixed in 2.9.8, please let me know and we will move it back.

alxy, 2024-06-19:
@marcschier We are experiencing this more and more especially with bigger subscription while beeing on `2.9.4`. We are gradually moving to `2.9.6` now, but I want to highlight the criticality of this issue. We basically lose huge amounts of data as soon as this happens, and for now we can only restart the publisher.

marcschier, 2024-06-19:
If you are losing data this is a different issue. What server are you using?  If you look at the logs, can you identify when publisher stops working?  Is this when `Publish #37889, Reconnecting=False, Error: BadTooManyPublishRequests` happens and onwards?  Or at the point of

```
2024-04-29T06:49:15.587Z [24-04-29 06:49:15.5877] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
2024-04-29T06:49:15.588Z       Got Keep Alive error: *BadNoCommunication 'Server not responding to keep alive requests.' (04/29/2024 06:49:15:BadNoCommunication 'Server not responding to keep alive requests.'
2024-04-29T06:49:15.588Z [24-04-29 06:49:15.5877] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
2024-04-29T06:49:15.588Z       Got Keep Alive error: BadNoCommunication 'Server not responding to keep alive requests.' (04/29/2024 06:49:15:BadNoCommunication 'Server not responding to keep alive requests.'
2024-04-29T06:49:15.622Z [24-04-29 06:49:15.6225] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
2024-04-29T06:49:15.622Z       Publish #191016, Reconnecting=False, Error: BadSessionIdInvalid
```
How many subscriptions are you creating on the session?  How many publish requests are inflight as per the diagnostic logs?

marcschier, 2024-06-19:
#2256

alxy, 2024-06-19:
> If you are losing data this is a different issue. What server are you using? If you look at the logs, can you identify when publisher stops working? Is this when `Publish #37889, Reconnecting=False, Error: BadTooManyPublishRequests` happens and onwards? Or at the point of
> 
> ```
> 2024-04-29T06:49:15.587Z [24-04-29 06:49:15.5877] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
> 2024-04-29T06:49:15.588Z       Got Keep Alive error: *BadNoCommunication 'Server not responding to keep alive requests.' (04/29/2024 06:49:15:BadNoCommunication 'Server not responding to keep alive requests.'
> 2024-04-29T06:49:15.588Z [24-04-29 06:49:15.5877] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
> 2024-04-29T06:49:15.588Z       Got Keep Alive error: BadNoCommunication 'Server not responding to keep alive requests.' (04/29/2024 06:49:15:BadNoCommunication 'Server not responding to keep alive requests.'
> 2024-04-29T06:49:15.622Z [24-04-29 06:49:15.6225] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
> 2024-04-29T06:49:15.622Z       Publish #191016, Reconnecting=False, Error: BadSessionIdInvalid
> ```
> 
> How many subscriptions are you creating on the session? How many publish requests are inflight as per the diagnostic logs? We might be a bit heavy handed on the publish request queueing side for the server you are using. Seems like the server just hangs up its side of the session since it feels it is not treated nicely :-).
> 
> The number of publish requests queued matches the number of subscriptions on the session. If there are 20 subscriptions we queue 20 publish requests, we ramp this down with every BadTooManyPublishRequests error we receive, up to the number of "publish requests queued and not errored" minus 1 (GoodPublishRequestCount - 1).
> 
> You can limit the number of publish requests using a ratio using `ppr|percentpublishrequests` command line option, e.g., if you have 40 subscriptions, setting it to 20 will only queue 8 requests, the default is 100%. While there is an option to declare the min number of requests that should always be queued (3), there is no option yet to limit the number of publish requests to a hard max value, but maybe that would make also sense.

I thinkt the behavior is exactly the same as in the original bug report. Reading nodes via the api is still possible, jsut the subscriptions silently breaks after a few days. Restarting publisher fixes the problem.
It's hard for me to pin-point the exact tiem when it happened, since I can only see the limited amount of logs in the publisher and once its broken it basically keeps repeating these lines:
```
[24-06-12 11:13:22.6501] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Subscription <<UnknownDataSet>>_($e3cbba8883fe746c6e35783c9404b4bc0c7ee9eb):41 STOPPED!
[24-06-12 11:13:22.6515] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #31988, Reconnecting=False, Error: BadSessionIdInvalid
[24-06-12 11:13:24.7867] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Got Keep Alive error: BadNoCommunication 'Server not responding to keep alive requests.' (06/12/2024 11:13:24:BadNoCommunication 'Server not responding to keep alive requests.'
[24-06-12 11:13:24.7868] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Got Keep Alive error: BadNoCommunication 'Server not responding to keep alive requests.' (06/12/2024 11:13:24:BadNoCommunication 'Server not responding to keep alive requests.'
[24-06-12 11:13:24.7870] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Could not send keep alive request: Opc.Ua.ServiceResultException BadConnectionClosed
[24-06-12 11:13:27.0325] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Subscription <<UnknownDataSet>>_($e3cbba8883fe746c6e35783c9404b4bc0c7ee9eb):10 STOPPED!
[24-06-12 11:13:27.0328] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Unexpected error sending publish request.
      Opc.Ua.ServiceResultException: Channel has been closed.
         at Opc.Ua.ClientBase.get_TransportChannel()
         at Opc.Ua.Client.Session.BeginPublish(Int32 timeout)
[24-06-12 11:13:27.6423] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Got Keep Alive error: BadNoCommunication 'Server not responding to keep alive requests.' (06/12/2024 11:13:27:BadNoCommunication 'Server not responding to keep alive requests.'
[24-06-12 11:13:27.6434] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Got Keep Alive error: BadSessionIdInvalid 'Session unavailable for keep alive requests.' (06/12/2024 11:13:27:BadSessionIdInvalid 'Session unavailable for keep alive requests.'
[24-06-12 11:13:27.8209] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Subscription <<UnknownDataSet>>_($e3cbba8883fe746c6e35783c9404b4bc0c7ee9eb):358736176 STOPPED!
[24-06-12 11:13:27.8212] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Unexpected error sending publish request.
      Opc.Ua.ServiceResultException: BadConnectionClosed
         at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.BeginSendRequest(IServiceRequest request, Int32 timeout, AsyncCallback callback, Object state)
         at Opc.Ua.Client.Session.BeginPublish(Int32 timeout)
```

I'm not exactly familiar with the internals of OPC UA, but our setup usually looks like this:
[... cut ...]
- There can be other processes as well of course that use these gateway and also create subscriptions on them. These systems don't have the problem we are experiencing with publisher. They likely use smaller subscriptions though.
- After a few days of operation, we noticed publisher stopped sending ANY data upstream and it only shows the error messages presented above

That beeing said we have no upgrade all sites that show the problem to `2.9.6` and I will monitor these deployments to see what happens, and if something happens try to store as many logs as possible.

closed 2024-06-20 by marcschier