Azure/Industrial-IoT #2224: Wrong template names shown in docs

RollsChris, 2024-05-18:
**Describe the bug**
Wrong template variables names used in documentation:

For example these do not work:
`{DataSetWriter}` 
`{WriterGroup}`

But these do:
{DataSetWriterGroup}
DataSetWriterId

**To Reproduce**
Look at docs and try for yourself

**Expected behavior**
To be correct

**Screenshots**
[image]

marcschier, 2024-06-07:
You were using an older version of OPC Publisher, but the help reflects current main line state, these are the new names in the doc corresponding to the latest version of OPC Publisher, but the older template names also work in newer versions of OPC Publisher.

closed 2024-06-07 by marcschier

RollsChris, 2024-06-10:
Hey we are using 2.9.6.4+978a207cf9 which also appears to be the latest?

marcschier, 2024-06-14:
Mmmmh.  Surprisingly, the end to end tests pass with the new and old variable names.  2.9.8 just released, let's re-open if the issue persists.