Azure/Industrial-IoT #2229: Add more Info to Logs "Too many publish request errer"

steffenbeermann, 2024-05-22:
**Is your feature request related to a problem? Please describe.**
We have configured multiple OPC Servers on the same Publisher module. When a server (or multiple) response "Too many publish request" it is unclear which connection caused it from the logs, for example:
`[24-05-22 12:41:18.3553] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
Publish #512, Reconnecting=False, Error: BadTooManyPublishRequests`

**Describe the solution you'd like**
Add more information to the logs like the subscription id, for example
`[24-05-22 12:41:18.3553] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
Publish #512, Reconnecting=False, Error: BadTooManyPublishRequests, SubscriptionId: XYZ`
**Describe alternatives you've considered**
Have a separate module with only one connection configured to troubleshoot which connection causes the issue

**Additional context**
None

steffenbeermann, 2024-05-23:
That would be totally fine as well. as long as we can assign it to one connection we would be more than happy!

marcschier, 2024-06-10:
I added the client session name to every client log message and also log all publish errors with the session name.

closed 2024-06-10 by marcschier

steffenbeermann, 2024-06-18:
Just tested it and works fine, thank you very much for the quick implementation!