Azure/Industrial-IoT #2231: OPC Publisher - Unhandled error BadTimeout during Publish

MattCosturos, 2024-05-29:
**Describe the bug**
OPC Publisher fails to publish any messages

**To Reproduce**
I am trying to use this module for the first time, so my setup is probably wrong. There are no example published_nodes.json files for 2.9, so I had to make up my own.

Steps to reproduce the behavior:
1. Deploy OPC Publisher module.
2. Configure opc publisher by editing the published_nodes.json file.
3. View module logs to see the publish fails.

Create Options from Deployment manifest
```
{
    "Hostname": "myHostName",
    "Cmd": [
        "-c",
        "--cl=5",
        "--PkiRootPath=/mount/pki",
        "--pf=/mount/published_nodes.json",
        "--lf=/mount/publisher.log",
        "--sl",
        "--ll=Debug",
        "--mm=PubSub",
        "--me=Json",
        "--fm=true",
        "--fd=false",
        "--bs=100",
        "--aa"
    ],
    "HostConfig": {
        "Binds": [
            "/path/to/my/files:/mount"
        ],
        "CapDrop": [
            "CHOWN",
            "SETUID"
        ]
    }
}
```

published_nodes.json
```
[
  {
    "EndpointUrl": "opc.tcp://my.server:1234",
    "DataSetWriterGroup": "ST38_114180",
    "DataSetWriterId": "ST38_114180_WriterId",
    "UseSecurity" : true,
    "OpcAuthenticationMode" : "UsernamePassword",
    "OpcAuthenticationUsername" : "myUsername",
    "OpcAuthenticationPassword" : "myPassword",
    "DataSetPublishingInterval" : "1000",
    "OpcNodes": [
      {
        "Id": "ns=2;s=ST38_114180.Path1.Controller_Mode",
        "OpcSamplingIntervalTimespan": "00:00:02",
        "OpcPublishingIntervalTimespan": "00:00:05",
        "HeartbeatIntervalTimespan": "00:01:00"
      },
      {
        "Id": "ns=2;s=ST38_114180.Path1.Feedrate_Override",
        "OpcSamplingIntervalTimespan": "00:00:02",
        "OpcPublishingIntervalTimespan": "00:00:05",
        "HeartbeatIntervalTimespan": "00:01:00"
      },
      {
        "Id" : "ns=2;s=ST38_114180.Path1.Rapid_Override",
        "OpcSamplingIntervalTimespan": "00:00:02",
[... 121 lines cut ...]
‼ production readiness: Edge Hub's storage directory is persisted on the host filesystem - Warning
    The edgeHub module is not configured to persist its /tmp/edgeHub directory on the host filesystem.
    Data might be lost if the module is deleted or updated.
    Please see https://aka.ms/iotedge-storage-host for best practices.
√ Agent image is valid and can be pulled from upstream - OK
√ proxy settings are consistent in aziot-edged, aziot-identityd, moby daemon and config.toml - OK

Connectivity checks
-------------------
√ container on the default network can connect to upstream AMQP port - OK
√ container on the default network can connect to upstream HTTPS / WebSockets port - OK
√ container on the IoT Edge module network can connect to upstream AMQP port - OK
√ container on the IoT Edge module network can connect to upstream HTTPS / WebSockets port - OK
28 check(s) succeeded.
7 check(s) raised warnings. Re-run with --verbose for more details.
2 check(s) were skipped due to errors from other checks. Re-run with --verbose for more details.
```

**Expected behavior**
I expect the opc messages to get published.

**Desktop (please complete the following information):**
- OS: Ubuntu 20.04.6 LTS
- IOT Edge runtime 1.4 series
- OPC Publisher 2.9.6

marcschier, 2024-05-30:
Timeout errors in log are not an issue, data seems to flow (3 value changes received, which are the initial values), since you use batch mode (--bs) the data will be sent onwards when the batch triggers.   Leave it running and if no data shows up, take a look at the troubleshoot.md file, but actually all looks good.

MattCosturos, 2024-05-30:
Data does flow into the module, When I check my kepserver instance, the client count increases / decreases as expected when I start / stop this module. 

I'm a bit confused how a timeout during publish could not be an issue. My understand is the OPC publisher module subscribes to the nodes that are configured, and gets changes from the opc server. Then eventually it will publish those changes out as an IOT message. I assumed the BadTimeout during publish was indicating the outgoing messages failed to send.

I was monitoring the built in endpoint and never saw any messages. Nor did my other edge module receive any opc messages.

The routes in my deployment for reference.
```
               "opc-publisher-1-to-upstream": {
                        "route": "FROM /messages/modules/opc-publisher-1/* INTO $upstream"
                    },
                    "metrics": {
                        "priority": 2,
                        "route": "FROM /messages/modules/metrics/outputs/metricOutput INTO $upstream"
                    },
                    "opc-publisher-1-output-to-star-machines-inputFromOpc": {
                        "route": "FROM /messages/modules/opc-publisher-1/* INTO BrokeredEndpoint(\"/modules/star-machines/inputs/inputFromOpc\")"
                    }
```

In the metrics summary, there are 0 Egress messages, does that stat indicate writes to the opc server, or iot messages going out?

I will test without batch mode, and look at the troubleshooting guide as well

MattCosturos, 2024-05-30:
Quick update to my previous comment.
I disabled batching `--bs=1` and `--bi=0` and now I see the Egress Message stats changing
```
 # Egress Messages queued/dropped     :             16 | 0
  # Egress Message send failures       :             33
  # Egress Messages successfully sent  :              0 
```

I'm also getting new errors, which is always fun.

```
  #47: Error 'One or more errors occurred. (Transient network error occurred, please retry.)' during sending network message. Retrying in 00:00:04.7000000...
      System.AggregateException: One or more errors occurred. (Transient network error occurred, please retry.)
       ---> Microsoft.Azure.Devices.Client.Exceptions.IotHubCommunicationException: Transient network error occurred, please retry.
       ---> System.Net.Sockets.SocketException (111): Connection refused
         at Microsoft.Azure.Amqp.ExceptionDispatcher.Throw(Exception exception)
         at Microsoft.Azure.Amqp.AsyncResult.End[TAsyncResult](IAsyncResult result)
         at Microsoft.Azure.Amqp.Transport.AmqpTransportInitiator.ConnectAsyncResult.End(IAsyncResult result)
         at Microsoft.Azure.Amqp.Transport.AmqpTransportInitiator.<>c.<ConnectAsync>b__17_1(IAsyncResult r)
         at System.Threading.Tasks.TaskFactory`1.FromAsyncCoreLogic(IAsyncResult iar, Func`2 endFunction, Action`1 endAction, Task`1 promise, Boolean requiresSynchronization)
      --- End of stack trace from previous location ---
         at Microsoft.Azure.Devices.Client.Transport.AmqpIot.AmqpIotTransport.InitializeAsync(CancellationToken cancellationToken)
         at Microsoft.Azure.Devices.Client.Transport.Amqp.AmqpIotConnector.OpenConnectionAsync(CancellationToken cancellationToken)
         at Microsoft.Azure.Devices.Client.Transport.Amqp.AmqpConnectionHolder.EnsureConnectionAsync(CancellationToken cancellationToken)
         at Microsoft.Azure.Devices.Client.Transport.Amqp.AmqpConnectionHolder.EnsureConnectionAsync(CancellationToken cancellationToken)
         at Microsoft.Azure.Devices.Client.Transport.Amqp.AmqpConnectionHolder.OpenSessionAsync(IDeviceIdentity deviceIdentity, CancellationToken cancellationToken)
         at Microsoft.Azure.Devices.Client.Transport.AmqpIot.AmqpUnit.EnsureSessionIsOpenAsync(CancellationToken cancellationToken)
         at Microsoft.Azure.Devices.Client.Transport.AmqpIot.AmqpUnit.EnsureSessionIsOpenAsync(CancellationToken cancellationToken)
         at Microsoft.Azure.Devices.Client.Transport.AmqpIot.AmqpUnit.OpenAsync(CancellationToken cancellationToken)
         at Microsoft.Azure.Devices.Client.Transport.Amqp.AmqpTransportHandler.OpenAsync(CancellationToken cancellationToken)
         at Microsoft.Azure.Devices.Client.Transport.ProtocolRoutingDelegatingHandler.OpenAsync(CancellationToken cancellationToken)
         at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.<>c__DisplayClass27_0.<<ExecuteWithErrorHandlingAsync>b__0>d.MoveNext()
      --- End of stack trace from previous location ---
         at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
         --- End of inner exception stack trace ---
         at Furly.Azure.IoT.Edge.Services.IoTEdgeHubSdkClient.ModuleClientAdapter.CreateAsync(String product, IotHubConnectionStringBuilder cs, String deviceId, String moduleId, ITransportSettings transportSetting, TimeSpan timeout, ILogger logger, IIoTEdgeClientState callback)
         at Furly.Azure.IoT.Edge.Services.IoTEdgeHubSdkClient.CreateAdapterAsync(IotHubConnectionStringBuilder cs, List`1 settings)
         --- End of inner exception stack trace ---
         at Furly.Azure.IoT.Edge.Services.IoTEdgeHubSdkClient.CreateAdapterAsync(IotHubConnectionStringBuilder cs, List`1 settings)
         at Furly.Azure.IoT.Edge.Services.IoTEdgeHubSdkClient.SendEventAsync(Message message, String output, CancellationToken ct)
         at Furly.Azure.IoT.Edge.Services.IoTEdgeEventClient.IoTEdgeMessage.SendAsync(CancellationToken ct)
         at Azure.IIoT.OpcUa.Publisher.Services.NetworkMessageSink.SendAsync(ValueTuple`2 message) in /home/vsts/work/1/s/src/Azure.IIoT.OpcUa.Publisher/src/Services/NetworkMessageSink.cs:line 250
```

There are no errors in edgeHub logs, and `iotedge check` shows all connections to the iothub are good for all protocols.

MattCosturos, 2024-05-31:
Final update.
Error was self inflicted.
In the Create Options for the OPC Publisher module, I had `"Hostname" : "myHostName"` which was set the literal hostname of the actual edge device. I am not sure where I found that, I may have mis-understood something my coworker was doing. Once I removed that from the create options the module started publishing messages correctly.