Azure/Industrial-IoT #2249: OPC Publisher not sending data continuously.

developedbymayur, 2024-06-14:
My IoT edge has 2 system modules , 2 Nodered and around 18 OPC publishers. Each OPC publisher has around 1700 Tags where 500 scan and publish at 1 second on value change and rest in One Minute scan and publish on Value Change. It was working well for Three years Until my OPCUA server which is Kepserver expired it Server and Client Driver certificate which was generated by SYSTEM while application was installed. Then as per PTC Kepserver we changed the certificate by reissuing it from application. We were used BasicSha256Sha Sign and Encrypt handshake .The certificates were imported in appropriate places but still issue persists.
   I have pasted logs if anyone can throw light on this issue. One observation is OPC publisher can be seen scanning messages but dos not send it ahead on route.

`

Option publisher wrong or not supported, please use -h option to get all the supported options.
[12:58:06.0320] warn: Microsoft.AspNetCore.DataProtection.Repositories.FileSystemXmlRepository[60]
      Storing keys in a directory '/root/.aspnet/DataProtection-Keys' that may not be persisted outside of the container. Protected data will be unavailable when container is destroyed.
[12:58:06.0568] info: Microsoft.AspNetCore.DataProtection.KeyManagement.XmlKeyManager[62]
      User profile is available. Using '/root/.aspnet/DataProtection-Keys' as key repository; keys will not be encrypted at rest.
[12:58:07.3641] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherModule[0]
      0: Module Line63 disconnected due to Communication_Error...
[12:58:07.4289] warn: Microsoft.AspNetCore.Server.Kestrel[0]
      Overriding address(es) 'http://+:80'. Binding to endpoints defined via IConfiguration and/or UseKestrel() instead.
[12:58:07.5016] info: Microsoft.Hosting.Lifetime[14]
      Now listening on: http://[::]:80
[12:58:07.5016] info: Microsoft.Hosting.Lifetime[14]
      Now listening on: https://0.0.0.0:443
[12:58:07.8155] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      Own certificate Subject 'CN=Azure.IIoT.OpcUa.Publisher.Module, C=DE, S=Bav, O=Microsoft, DC=af09:63e3:abcf:2582:ad75:3e13:13aa:ba1e' (Thumbprint: 6244530753191F708DE525F6057B198FC02A0E54) loaded.
[12:58:07.9988] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      Application own certificate store contains 1 certs.
[12:58:08.0004] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      01: Subject 'CN=Azure.IIoT.OpcUa.Publisher.Module, C=DE, S=Bav, O=Microsoft, DC=af09:63e3:abcf:2582:ad75:3e13:13aa:ba1e' (Thumbprint: 6244530753191F708DE525F6057B198FC02A0E54)
[12:58:08.0007] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      Trusted issuer store contains 0 certs.
[12:58:08.0016] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      Trusted issuer store has 0 CRLs.
[12:58:08.0024] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      Trusted peer store contains 1 certs.
[12:58:08.0025] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      01: Subject 'CN=Azure.IIoT.OpcUa.Publisher.Module, C=DE, S=Bav, O=Microsoft, DC=af09:63e3:abcf:2582:ad75:3e13:13aa:ba1e' (Thumbprint: 6244530753191F708DE525F6057B198FC02A0E54)
[12:58:08.0026] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      Trusted peer store has 0 CRLs.
[12:58:08.0027] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      Rejected certificate store contains 0 certs.
[12:58:08.0706] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherConfigurationService[0]
      Found published Nodes File with hash B9B4CB63F6B3F9D8B27604D1B23944F764ABAFA372F3CA3480531BFF06BC5E4E, loading...
[12:58:08.1787] fail: Azure.IIoT.OpcUa.Publisher.Services.PublisherConfigurationService[0]
      SerializerException while loading job from file.
[12:58:08.1978] info: Azure.IIoT.OpcUa.Publisher.Discovery.NetworkDiscovery[0]
      Starting discovery processor...
[12:58:09.5781] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherModule[0]
      0: Module Line63 reconnected due to Connection_Ok.
[12:58:09.5876] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherModule[0]
      0: Module Line63 opened.
[12:58:10.4367] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherModule[0]
      Starting OpcPublisher module version 2.9.0.1...
[12:58:10.4440] info: Azure.IIoT.OpcUa.Publisher.Services.RuntimeStateReporter[0]
      Api Key exists in IoTEdge store...
[12:58:10.4441] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherModule[0]
      OpcPublisher module version 2.9.0.1 started.
[12:58:10.4460] info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
[12:58:10.4461] info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
[12:58:10.4461] info: Microsoft.Hosting.Lifetime[0]
      Content root path: /app
[13:02:15.8574] info: Microsoft.Hosting.Lifetime[0]
      Application is shutting down...
[13:02:15.8607] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherModule[0]
      Stopped module OpcPublisher.
[13:02:15.8847] info: Azure.IIoT.OpcUa.Publisher.Discovery.NetworkDiscovery[0]
      Cancelling all pending requests...
[13:02:15.8848] info: Azure.IIoT.OpcUa.Publisher.Discovery.NetworkDiscovery[0]
      Pending requests cancelled...
[13:02:15.8848] info: Azure.IIoT.OpcUa.Publisher.Discovery.NetworkDiscovery[0]
      Stopped discovery processor.
[13:02:15.9221] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      Stopping all clients...
[... 384 lines cut ...]
  DIAGNOSTICS INFORMATION for          : Line63 (OPC Publisher 2.9.0.1)
  # Time                               : 2024-06-13T14:54:07.6491130Z
  # Ingestion duration                 :    00:01:36:53 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :             99 (All time ~0.02/s; 01 in last 60s ~0.02/s)
  # Ingress EventData (from OPC)       :              0
  # Ingress ValueChanges (from OPC)    :         11,649 (All time ~2/s; 01 in last 60s ~1.88/s)
  # Ingress Events (from OPC)          :              0
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :             99
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :             99
  # Encoder avg Notifications/Message  :              1
  # Encoder worst Message split ratio  :              0
  # Encoder avg IoT Message body size  :         23,524
  # Encoder avg IoT Chunk (4 KB) usage :            5.7
  # Estimated IoT Chunks (4 KB) per day:              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :             99 (0.02/s)
  # Connection retries                 :              1
  # Opc endpoint connected?            :           True
  # Monitored Opc nodes succeeded count:           1025
  # Monitored Opc nodes failed count   :              0
`

marcschier, 2024-06-14:
Please use version 2.9.8 , 2.9.0 is not supported anymore.  The issues you are seeing are on the northbound to EdgeHub, the connection to Kepserver looks good:

`Successfully created subscription opc.tcp://xx.xx.xx.xx:49320_A11D0D4A_Line63:BooleanTags:0'.`

We saw errors like these in the past, but not recently, so it was likely due to a SDK issue or IoT edge issue, and it has since been fixed (and 2.9.8 uses the latest IoT SDKs).  If you see issues with 2.9.8, please re-open (only need to send log for the error you see, no need to paste the full logs).

developedbymayur, 2024-06-15:
I updated it to 2.9.8 but behaviour is same.
Every 1 hour im getting this error : 

[24-06-14 11:09:42.2557] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherModule[0]
      1: Module Line61 disconnected due to Communication_Error...
[24-06-14 11:09:55.4708] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherModule[0]
      1: Module Line61 reconnected due to Connection_Ok.   

Every 45 mins : 
[24-06-14 14:35:08.3151] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      Accepting untrusted peer certificate 81AAA66B8F9EA9E3E1BEAFCD2065354687F50330, 'CN=KEPServerEX/UA Server, O=Unknown, C=US, DC=ACPLDIIOTSRV01.ACGWORLD.COM' due to AutoAccept(UntrustedCertificates) set!

marcschier, 2024-06-18:
This is expected, the OPC UA connection exchanges a refresh token every n duration, negotiated with the server.  The disconnected due to Communication_Error and Connection_OK are equally expected, MQTT does not support inline exchange of auth secrets.  If you can send logs of publisher starting or from the point where no data is flowing anymore, I will have a look.  One doc that might be helpful is the troubleshooting.md guide.  If that does not work, some command line options that might be helpful are `--ln`, `--len` and `--ll=Debug` to get more data from the logs.  If you have a MSFT support contact, you can also open a ticket and supply supportbundle, and it will end with me to dig deeper.

developedbymayur, 2024-06-20:
The OPCUA certificate was not generated with full privileges from Server. Thus causing OPC publisher to trust the Server certificate.

marcschier, 2024-06-28:
Handling this on a side channel.

closed 2024-06-28 by marcschier