Azure/Industrial-IoT #2264: Subscription goes stale after operating for a few days

alxy, 2024-06-25:
**Describe the bug**
This is a follow up of #2215
We have upgraded our publisher to `2.9.6` in the meantime, and we still the the same behavior. We use MQTT transport. What happens is after a couple of days of successful operation, the publisher doesn't send any data upstream for a given writer group. Other writer groups are not affected (that are configured for the same server)

**To Reproduce**
1. Setup publisher to use MQTT transport (not sure if relevant), and subscribe to a large amount of nodes (in this case, 129 nodes)
2. Let publisher run for some time (in our case, some days)
3. After some time, the publisher stops working - however, the logs do not really indicated a problem apart from the `BadTooManyPublishRequests`:
 ```
[24-06-25 11:50:28.5091] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #126788, Reconnecting=False, Error: BadTooManyPublishRequests
[24-06-25 11:50:28.5093] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Too many publish request error: Limiting number of requests to 5...

  DIAGNOSTICS INFORMATION for          : opc-cz-ok-2_Kepware
  # OPC Publisher Version (Runtime)    : 2.9.6.4+978a207cf9 (.NET 8.0.4/linux-x64/OPC Stack 1.5.374.36)
  # Time                               : 2024-06-25T11:50:32.1715136Z
  # Ingestion duration                 :    06:02:35:26 (dd:hh:mm:ss)
  # Endpoints connected/disconnected   :              1 | 0 (Connected)
  # Connection retries                 :              3
  # Subscriptions count                :              5
  # Good/Bad Monitored Items count     :            128 | 1
  # Queued/Minimum request count       :              5 | 5
  # Good/Bad Publish request count     :              5 | 0
  # Ingress value changes              :        117,806 (All time ~0.22/s; 0 in last 60s ~0/s)
  # Ingress events                     :              0 
  # Ingress values/events unassignable :              0
  # Server queue overflows             :              5 (All time ~0/s; 0 in last 60s ~0/s)
  # Received Data Change Notifications :         36,227 (All time ~0.07/s; 0 in last 60s ~0/s)
  # Received Event Notifications       :              0 
  # Received Keep Alive Notifications  :        166,150 (All time ~18.89/min)
  # Generated Cyclic read Notifications:              0 
  # Generated Heartbeat Notifications  :              0 
  # Generated Model Changes            :              0 
  # Notification batch buffer size     :              0
  # Encoder input/output buffer size   :              0 | 0
  # Encoder Notif. processed/dropped   :         36,227 | 0
  # Encoder Network Messages produced  :         34,884
  # Encoder avg Notifications/Message  :              1
  # Encoder worst Message split ratio  :              0
  # Encoder avg Message body size      :          1,817
  # Encoder avg Chunk (4 KB) usage     :            0.4
  # Estimated Chunks (4 KB) per day    :              0
  # Egress Messages queued/dropped     :              0 | 0
  # Egress Message send failures       :              0
  # Egress Messages successfully sent  :         34,884 (0.07/s)
```
4. The `Egress Messages successfully sent` and `Ingress value changes` basically stay the same once this happens

**Expected behavior**
OPC publisher should continue to function. We have other subscriptions on the same server that continue to work. We have even connected another publisher instance with a smaller subscription and that one is still producing data.

**Additional context**
OPC Publisher Version (Runtime)    : 2.9.6.4+978a207cf9 (.NET 8.0.4/linux-x64/OPC Stack 1.5.374.36)

marcschier, 2024-06-25:
Could you try the `2.9.9-preview2` tag?  Will be released later this week.  Feedback before we release would be greatly appreciated.

marcschier, 2024-06-25:
In addition to a lot of fixes :-) - there is new option to set an upper limit for the number of publish requests (`--xpr`).

alxy, 2024-06-26:
Yep, we will do it, but it will take time again. Which fix/commit do you think will resolve the problem? I've also seen that this (`BadTooManyPublishRequests`) is actually a server error message. But my understanding is that after some time the server should recover from that and start sending some data again. For us, it's not the case.

marcschier, 2024-06-26:
No problem.  We are releasing 2.9.9 today. The theory on this is that we are keeping too many publish requests that the server is not able to handle, so we did 2 things: a) introduce a cap on publish requests, b) allowing that cap to be overridden using command line, c) calculating the publish requests to submit as "good -1" instead of "good", so there is one less than what we believe is currently "working".  The calculation is done [here](https://github.com/Azure/Industrial-IoT/blob/b049fbaf57104aa4344d278b2deeb30daf61eaa4/src/Azure.IIoT.OpcUa.Publisher/src/Stack/Services/OpcUaClient.cs#L1320) and [here](https://github.com/Azure/Industrial-IoT/blob/b049fbaf57104aa4344d278b2deeb30daf61eaa4/src/Azure.IIoT.OpcUa.Publisher/src/Stack/Services/OpcUaClient.cs#L1116).  

What we are seeing here might or might not be related to the publish requests.  It looks from the picture that a lot of keep alive notifications are coming in (which are sent by server when no value changes are reported).  If that ticks up but no data is coming, the subscription is in working state, the issue then is on the server.  

Experimental addition in 2.9.9 - monitored item watchdog.  If you expect items should report at least every, let's say 1 hour minimum, set the watchdog to 1 hour, and the watchdog behavior to "reset" and that will close the subscription and reestablish if no data is received on any monitored item (nodeId) within this 1 hour.  More aggressive behavior can be set to Exit/FailFast to crash the publisher and let docker restart.  Or less aggressive is the default which just logs the watchdog failure as metric/diagnostic output.

marcschier, 2024-06-28:
If you see the issue with 2.9.9 (now released), please could you update this issue?  We might then need pcap traces to see what goes over the wire when it stops.  In the deploy/docker folder is a compose file that shows how to get those using a "side car" like approach.  Or you use wireshark directly of course. 

Also, in the logs you will find lines which are something like "Subscription stopped", "Subscription recovered" would be interested if the subscription is stopped and never recovered.

alxy, 2024-07-09:
To give an update here: We are still running on `2.9.6` right now, but we have decreased the publishing intervals a bit (clients have set the to `1 second` for basically all observed signals). Now it looks much better and the setup is already running now for ~2 weeks. We will continue to observe this and in mid-future also upgrade to latest version to get more elaborate logs when/why exactly this happens.

closed 2024-07-09 by marcschier