Azure/Industrial-IoT #2277: Set Timestamp in UserProperties of message when using MQTT Transport

quality-leftovers, 2024-07-02:
**Is your feature request related to a problem? Please describe.**
When receiving messages locally via MQTT and processing or enhancing them it would be great if we could lookup the timestamp when the publisher created the message in the user properties of the message instead of having to parse the complete message.

In other Transports like RabbitMQ and DAPR the Timestamp is set.

**Describe the solution you'd like**
Set the Property "Timestamp" in MQTT user properties (either when EnableRoutingInfo is set, or always). 
Currently the timestamp is set on the `IEvent`, but ignored when using MQTT as transport. See:

https://github.com/microsoft/project-furly/blob/01480644d1b985c49b6b820db6179ef316ad79cd/src/Furly.Extensions.Mqtt/src/Clients/MqttMessage.cs#L125-L129

**Describe alternatives you've considered**
- Having a microservice consume all messages, add the timestamp and re-publish them to another topic
- Using DAPRD as middleware since it adds the Timestamp property. But DAPR trasnsport is already used to send the data to azure cloud  

**Additional context**
Add any other context or screenshots about the feature request here.

marcschier, 2024-07-02:
Easy to add, will do it in next release.

marcschier, 2024-07-03:
Added "TineStamp" property, but it is only supported with Mqtt v5

closed 2024-07-03 by marcschier

quality-leftovers, 2024-07-03:
Thanks ;).

That happend faster than I could blink.