Azure/Industrial-IoT #2285: Documentation/Clarification on using the new writer API

quality-leftovers, 2024-07-04:
The documentation states that the configuration should either happen using configuration or writer endpoint.

However I cannot see any "writer" API to get a list of all writers. Should we still use `/v2/configuration` for that?

GET /v2/write/ returns method not allowed.

marcschier, 2024-07-04:
Are you building/testing main branch?  The doc is [here](https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/api.md#writer).

marcschier, 2024-07-05:
One more info, the new API should not be used with the old API, or, more concretely, the old API allows creating entries where multiple entries can have the same writer if and group of, but different other settings.  The new APIs will then fail with an error that no unique entry to operate on can be identified.  Just to be aware.

quality-leftovers, 2024-07-10:
Let me clarify my question a bit, I think I didn't do a good job in the initial post :).

Basically I was just wondering whether `GET /v2/configuration` will be superseded by something else, too or if it's okay to use both - `GET /v2/configuration` (to get an overview) 
- and the writer API for configuration.

I just want to make sure we don't do something wrong when we switch to the new API.

Cheers

marcschier, 2024-07-10:
No, you can use all GET API calls just fine.  And all API calls are supported.  The writer API just operates on unique writer entries, and guarantees that, while the other API such as Set configuration or AddOrUpdateNodes can result in entries being created in the config that have the same group/writer id but different configuration, e.g. publishing interval etc.  In that case the new API can report an error.

closed 2024-07-10 by marcschier