Azure/Industrial-IoT #2287: Cannot connect to OPC-UA server anymore after have updated OPC Publisher from v2.9.4 to v2.9.9

jacqueskang, 2024-07-05:
**Describe the bug**

After have updated OPC Publisher from v2.9.4 to v2.9.9 it cannot connect to OPC-UA server anymore with error:

 > **Server did not return a Certificate matching the ApplicationUri specified in the EndpointDescription....**

I cannot find in release note any related breaking change.

**To Reproduce**

1. Run OPC Publisher v2.9.4 as Azure IoT Edge module
2. Configure OPC nodes as below
    ```json
    [
      {
        "EndpointUrl": "opc.tcp://wwvcamii0043.dc.ege.ds:49320",
        "UseSecurity": true,
        "OpcAuthenticationMode": "UsernamePassword",
        "OpcAuthenticationUsername": "#{OPC_SERVER_USERNAME}#",
        "OpcAuthenticationPassword": "#{OPC_SERVER_PASSWORD}#",
        "OpcNodes": [
           ...
        ]
      }
    ]
    ```
3. By inspecting logs we can see OPC Publisher is able to connect to our server:

```
[24-07-05 15:27:43.7072] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Selecting endpoint opc.tcp://wwvcamii0043.dc.ege.ds:49320/ with SecurityMode SignAndEncrypt and any SecurityPolicyUri from:
            #051: opc.tcp://wwvcamii0043.dc.ege.ds:49320/|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256]
[24-07-05 15:27:43.7086] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Endpoint #051: opc.tcp://wwvcamii0043.dc.ege.ds:49320/|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256] selected!
[24-07-05 15:27:44.1796] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #1: Creating session opc.tcp://wwvcamii0043.dc.ege.ds:49320_EEE23BBD_x with endpoint opc.tcp://wwvcamii0043.dc.ege.ds:49320/...
[24-07-05 15:27:44.7252] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session opc.tcp://wwvcamii0043.dc.ege.ds:49320_EEE23BBD_x with opc.tcp://wwvcamii0043.dc.ege.ds:49320 changed from Connecting to Ready
[24-07-05 15:27:44.7275] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      New Session opc.tcp://wwvcamii0043.dc.ege.ds:49320_EEE23BBD_x created with endpoint opc.tcp://wwvcamii0043.dc.ege.ds:49320/ (opc.tcp://wwvcamii0043.dc.ege.ds:49320).
[24-07-05 15:27:44.7276] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Client opc.tcp://wwvcamii0043.dc.ege.ds:49320_EEE23BBD_x [state:Ready|refs:30] CONNECTED to opc.tcp://wwvcamii0043.dc.ege.ds:49320/!
```
4. Update OPC Publisher to v2.9.9 with exactly the same configuration, connection is no longer possible.

```
[24-07-05 15:20:18.2191] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://wwvcamii0043.dc.ege.ds:49320_FD8BFF0B_x [state:Connecting|refs:30]: Discovery endpoint opc.tcp://wwvcamii0043.dc.ege.ds:49320/ returned endpoints. Selecting endpoint opc.tcp://wwvcamii0043.dc.ege.ds:49320/ with SecurityMode NotNone and any SecurityPolicyUri from:
            #051: opc.tcp://wwvcamii0043.dc.ege.ds:49320/|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256]
[24-07-05 15:20:18.2191] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://wwvcamii0043.dc.ege.ds:49320_FD8BFF0B_x [state:Connecting|refs:30]: Endpoint #051: opc.tcp://wwvcamii0043.dc.ege.ds:49320/|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256] selected!
[24-07-05 15:20:18.7281] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #1 - opc.tcp://wwvcamii0043.dc.ege.ds:49320_FD8BFF0B_x [state:Connecting|refs:30]: Creating session opc.tcp://wwvcamii0043.dc.ege.ds:49320_FD8BFF0B_x with endpoint opc.tcp://wwvcamii0043.dc.ege.ds:49320/...
[24-07-05 15:20:18.7367] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSession[0]
      : Session disposed.
[24-07-05 15:20:18.7369] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://wwvcamii0043.dc.ege.ds:49320_FD8BFF0B_x [state:NoTrust|refs:30]: Session opc.tcp://wwvcamii0043.dc.ege.ds:49320_FD8BFF0B_x with opc.tcp://wwvcamii0043.dc.ege.ds:49320 changed from Connecting to NoTrust
[24-07-05 15:20:18.7369] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #2 - opc.tcp://wwvcamii0043.dc.ege.ds:49320_FD8BFF0B_x [state:NoTrust|refs:30]: Failed to connect to opc.tcp://wwvcamii0043.dc.ege.ds:49320/: Server did not return a Certificate matching the ApplicationUri specified in the EndpointDescription....
[24-07-05 15:20:18.7370] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://wwvcamii0043.dc.ege.ds:49320_FD8BFF0B_x [state:NoTrust|refs:30]: Retrying connecting session in 00:00:00.5000000...
```

**Expected behavior**

Being able to update from v2.9.4 to v2.9.9 without error.

marcschier, 2024-07-05:
This was noted in the release notes.  It is due to a bug fix.  The issue was that customers could not update from 2.8 because the own certificate was regenerated.  We are aware that the fix breaks updating from versions that brole the behavior, but we are in a rock and a hard place.  Apologies.  

One option is to set the --cfa option to configure publisher from an existing valid certificate, then we read the cert, set the application Uri and other settings and that will force the stack to select the existing cert.  So to try, can you run with that flag set?  Let me know, maybe this is a complete different issue.

marcschier, 2024-07-05:
Although here the server returns something we don't expect, so could it also be that the server certificate is configured incorrectly and we might see a security fix in the UA stack we are using?

marcschier, 2024-07-05:
@mregen to take a look

marcschier, 2024-07-05:
Found the change: https://github.com/OPCFoundation/UA-.NETStandard/commit/f081d51bba719dda0fc2e94b268e8e976888143f.

Looks like this is to behave per spec. Looks like the server is the culprit, what server are you using?  Possible to open a ticket to them?

closed 2024-07-05 by marcschier

jacqueskang, 2024-07-08:
@marcschier thanks for information.

We are using KEPServerEX which a custom server certificate.
If I understand correctly the server certificate must have URI SAN matching the EndpointUrl specified in OPC Publisher's published nodes configuration, right?

marcschier, 2024-07-08:
The servers **application Uri** which uniquely identifies the kepserver installation and that the server presents during session create/activate against an endpoint url must also be in the accompanying certificate's SAN of that endpoint, the endpoint url host name is matched via the domain name in the cert (also in SAN).