Azure/Industrial-IoT #2289: The key / environment variable MaxPublishRequests does not work as expected

dimtix, 2024-07-12:
**Describe the bug**
The key / environment variable MaxPublishRequests seems to have no impact on the publisher behaviour.

**To Reproduce**
1. Start publisher via docker compose without this variable (default value) and with diffent values of this variable
2. Configure publisher with appropriate configuration 
4. Check the value of CurrentPublishRequestsInQueue for the publisher session in the server with any client (e.g. UaExpert)
5. The value of CurrentPublishRequestsInQueue is allways equal to the number of created subscriptions independent of the value of MaxPublishRequests 

**Expected behavior**
CurrentPublishRequestsInQueue == MaxPublishRequests 

**Screenshots**
**1**
[image]
[image]

**2**
[image]
[image]

**3**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser chrome
 - Version 2.9.9

marcschier, 2024-07-20:
Unfortunately this also requires a change in the UA stack, which overrides the max setting and always at least submits subscription count publish requests.  Hence this will take a bit of time.

marcschier, 2024-10-03:
#2352

closed 2024-10-03 by marcschier