Azure/Industrial-IoT #2293: Allow deleting all writers with a single DELETE v2/writer

quality-leftovers, 2024-07-18:
**Is your feature request related to a problem? Please describe.**
After updating to the new writer API my "app" tried deleting an writer using the writers API.
At that point my "app" didn't really know which writers were created using the old and which the new API.

When trying to delete a writer I got a
`{"type":"BadRequestException","title":"The request contains invalid information or the parameters of the operation are invalid.","status":400,"detail":"Field ids in writer entry must be present and unique."}` error. Probably because it was created the old API.

I used publisher version `b85f604` from 3th july. Need to check if it is still an issue with current version.  

**Describe the solution you'd like**
- allow deleting writers created using the old API via writers ap
- provide an endpoint in the writers API to "clear" everyhing to enable easy transition to writers API

**Describe alternatives you've considered**
none

**Additional context**
none

marcschier, 2024-07-19:
Bullet point 2:  Removing the entire configuration can be achieved using the [SetConfiguredEndpoints](https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/api.md#setconfiguredendpoints) API passing a [request](https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/definitions.md#setconfiguredendpointsrequestmodel) with an empty endpoints array.

I could add a flag to the delete to allow ambiguity and remove all entries with the specified writer and group.  Something like "?force=true" or so.

quality-leftovers, 2024-07-22:
A force option would be great.

So I assume deleting a set of writers and keeping some using the old API would be okay, too and just updating / inserting new writers using both the old and new API is disallowed?

marcschier, 2024-07-22:
Yes, that would be ok, provided you can manage the difference between which were added with new vs. old API, e.g., it would be perfectly fine to create with new API and then add a same named writer with different configuration using the old API, which then would fail a call on that writer with the new API.  I would therefore suggest to use either one or the other but not both, but there is nothing enforced in the code.

marcschier, 2024-07-22:
API has a new force flag, and ?force=true now allows removing all entries with group/writer.

closed 2024-07-22 by marcschier