Azure/Industrial-IoT #2310: When using samples mode, skip first notification after a session reconnect

marcschier, 2024-07-31:


DJmagalhaes, 2024-08-14:
Whenever a PLC gets disconnected / switched off and later the this device gets reconnected the OPC Publisher should not re-send the message which has already been sent to the IoTHub what should be provided based on the SourceTimestamp, Value and State (respecting the DataChangeTrigger setting). In our case we use PubSub format but I do not think it has any significance from the perspective of this particular issue (in the title of the github issue, you linked below, sample mode is mentioned).

marcschier, 2024-08-15:
As per discussion, downstream consumers should have resiliency to handle duplicate messages and duplicate values, this is not confined to just OPC Publisher side, but end to end possible in a distributed system.

closed 2024-08-15 by marcschier