Azure/Industrial-IoT #2312: Browsing Nodes does not work reliably

kappa-lhirsch, 2024-08-01:
**Describe the bug**
Using WebApi V2.9.10 and Publisher-Module V2.9.10
When I try to browse a node with the WebApi I cannot get a reliable response.

When I make a POST to `https://{{OPC-SERVICEURL}}/twin/v2/browse/{{endpointId}}`
With the payload:
```json
{
    "header": {
        "elevation": {
            "type": "UserName",
            "value": {
                "user": "{{user}}",
                "password": "{{password}}"
            }
        },
        "namespaceFormat": "Uri"
    }
}
```

I always get different responses:

### 1. Status 410 Gone
```json
{
    "Type": "OperationCanceledException",
    "Title": "The operation was cancelled by the system or due to user action.",
    "Status": 410,
    "Detail": "Reason unknown",
    "Instance": null,
    "Extensions": {}
}
```

### 2. Status 500 Internal Server Error
```json
{
    "Type": "ServiceResultException",
    "Title": "Unknown",
    "Status": 500,
    "Detail": "-2146233088",
    "Instance": null,
    "Extensions": {}
}
```

### 3. Status 200 Ok
The desired Output

To me this makes no sense at all, maybe timeouts are hit in the browsing process?
I wasn't able to find a timeout that is used for browsing, and all others are set to someting like 1 minute, so that shouldn't be the issue here.

Time to get a response is always arround 3 to 7 seconds.
I can tell by the logs that the session is closed after every browse request.

Since there is no recursive browsing feature, I have to make multiple request to the endpoint, but when a single request takes 5 seconds and fails every second time for no good reason, this is pretty inefficient.

**To Reproduce**
Steps to reproduce the behavior:
1. See above

**Expected behavior**
It works every time, since it is the same request every time.

marcschier, 2024-08-02:
What command line options are you setting?

marcschier, 2024-08-08:
@kappa-lhirsch - I tried to reproduce your issues with mcr.microsoft.com/iotedge/opc-publisher:2.9.11-preview1 and could not.  However, I ran into 1 timeout condition at the IoT Hub side and noticed that the default timeout for the cloud to device method invocation was 10 seconds.  I will up this to 60 seconds.  This could be the reason for the Gone status.

I used the industrial-iot cli browse command which has a -r option to browse recursively.  I start a publisher and the test server included in the repo, the CLI project, and the web API service is running in Azure App Service. Then select an endpoint and call `nodes browse -i <endpointId> -r` and I do not see any errors browsing ca 20000 nodes.  

If you need to keep the session active between browse calls, you must set a client linger timeout using `--cl` (in seconds) e.g. `--cl=15`.  5 seconds is the default configuration in the automatically deployed OPC Pubilsher layered deployment, but if you are deploying OPC Publisher yourself you might not have set it.  

Why it worked likely better in 2.9.2 is that we used to have the connect API which maintained a connection forever until disconnect, which resulted in several bugs and unexpected behavior so based on that we removed and solely rely on client linger (timing out sessions after no activity).

marcschier, 2024-08-08:
The service result error 0x80131500 seems like an extended Bad_SecurityChecksFailed error which happens when establishing connection to server.  This is the server reporting that the provided certificate might be bad.  If you can establish a lingering session it might happen less often, but if you find the reason on your server please let me know.

closed 2024-08-08 by marcschier