Azure/Industrial-IoT #2313: Heartbeat publishing wrong values at wrong time

kappa-lhirsch, 2024-08-01:
**Describe the bug**
Using WebApi V2.9.10 and Publisher-Module V2.9.10
When using a heartbeat interval on a node, to publish values even when no new values are available, I get a weired behavior.
I'm using PeriodicLKG  HeartbeatBehavior.

### Case 1
Setup:
* Node1, PublishingInterval, SampleInterval and HeartbeatInterval all set to 30 seconds, Value never updated
* Node2, PublishingInterval, SampleInterval set to 30 seconds, no HeartbeatInterval, Value is updated every second

[image]

When the Heartbeat of Node1 happens at the same time as Node2 get published, the value of Node2 becomes automatically 1. Afterwards the Heartbeat of Node1 doesn't happen at the same time as the others get published and the behavior is as expected.

### Case 2
Setup:
* Node1, PublishingInterval, SampleInterval and HeartbeatInterval all set to 30 seconds, Value never updated
* Node2, PublishingInterval, SampleInterval and HeartbeatInterval all set to 30 seconds, Value is updated every second

[image]

Now that Node2 also has a HeartbeatInterval set, every HeartbeatInterval an additional value, again 1, from Node2 is published.

### Case 3
Setup:
* Node1, PublishingInterval, SampleInterval set to 30 seconds, no HeartbeatInterval, Value never updated
* Node2, PublishingInterval, SampleInterval and HeartbeatInterval all set to 30 seconds, Value is updated every second

[image]

Now the Heartbeat-Value isn't 1 anymore and it is the last published value, which is correct. But why do I even get a Heartbeat when the last published value was only 20 seconds ago and the HeartbeatInterval is set to 30 seconds.

This whole Heartbeat thing is acting completely unreliable. Why does one value influence the other value to become 1 and why do I even get Heartbeats from the other value.

**To Reproduce**
Steps to reproduce the behavior:
1. See above

**Expected behavior**
Heartbeats are only published when no new value is available. Nodes shouldn't influence other nodes.

marcschier, 2024-08-02:
Can you provide sample configuration Json for case1,2,3?  Also, can you provide info what you are expecting?  Especially about PeriodicLKG behavior?

marcschier, 2024-08-07:
@kappa-lhirsch , could you try with mcr.microsoft.com/iotedge/opc-publisher:2.9.11-preview1 and let me know if this is resolved for you?` If not, please see my previous comment.

closed 2024-08-08 by marcschier