Azure/Industrial-IoT #2324: Please explain how to deploy the configuration file, including Name and Path.

uriel-kluk, 2024-08-26:
Please explain how to deploy the configuration file, including file Name and Path.

marcschier, 2024-08-26:
I am not sure what is the ask here.  the configuration file is provided to publisher using the --pf command line option.  If you are using docker, add a docker volume mount (see readme, examples, samples etc.) and the file in there, and reference via --pf, see other command line options in the help.

uriel-kluk, 2024-08-26:
Thanks Marc,

The comprehensive documentation wasn't very intuitive for me as a first-time user of this project, but your references to command-line options and environment variables clarified the issue.

Could I recommend to help developers to extend the documentation like this:

OPC Publisher Configuration
The simplest way to configure OPC Publisher is via a configuration file. A basic configuration file looks like this:

[
  {
    "EndpointUrl": "opc.tcp://testserver:62541/Quickstarts/ReferenceServer",
    "UseSecurity": true,
    "OpcNodes": [
      {
        "Id": "i=2258",
        "OpcSamplingInterval": 2000,
        "OpcPublishingInterval": 5000,
        "DisplayName": "Current time"
      }
    ]
  }
]

This configuration can be placed in a JSON file, typically named publishednodes.json, and provided to OPC Publisher at runtime.
Configuring via Command Line
In addition to using a configuration file, OPC Publisher can be configured via command-line options. This is useful for overriding settings or quickly testing changes. Some common command-line parameters include:

  *   --pf=<path>: Specifies the path to the published nodes configuration file.
  *   --di: Enables diagnostics output.
  *   --pn=<port>: Specifies the port on which the publisher listens for incoming telemetry.

Example usage:

iotedgehubdev start -d deployment.json --pf=/app/publishednodes.json --di

Configuring via Environment Variables
Environment variables can also be used to configure OPC Publisher. This method is particularly useful when deploying at scale or in environments where you want to externalize configuration from the container image.
Some environment variables you might use include:

  *   PUBLISHERCONFIGURATIONFILE: Path to the configuration file.
  *   OPCPUBLISHER_TELEMETRY_PORT: Sets the telemetry port.
  *   OPCPUBLISHER_LOGLEVEL: Configures the log level.

Example in a Docker docker-compose.yml file:

yaml
services:
  opcpublisher:
    image: mcr.microsoft.com/iotedge/opc-publisher
    environment:
      - PUBLISHERCONFIGURATIONFILE=/app/publishednodes.json
      - OPCPUBLISHER_LOGLEVEL=info
      - OPCPUBLISHER_TELEMETRY_PORT=62222
    volumes:
      - ./publishednodes.json:/app/publishednodes.json

Automating Deployment at Scale
For large-scale deployments, automating the configuration and management of OPC Publisher is critical. This can be achieved using Azure IoT Hub's Cloud-to-Device (C2D) messaging.
C2D commands allow you to remotely configure and control OPC Publisher instances running on IoT Edge devices. For example, you can send commands to update the configuration, restart the module, or change runtime parameters without needing to manually intervene on each device.
Example of sending a C2D command to update configuration:

bash
az iot hub invoke-module-method --hub-name <your-iot-hub> --device-id <your-device-id> --module-name <opc-publisher> --method-name UpdateConfiguration --method-payload '{"OpcNodes": [{"EndpointUrl": "opc.tcp://new-opc-server:4840", "OpcNodes": [{"Id": "ns=2;i=10853"}]}]}'

This method allows you to dynamically adjust the configuration across multiple devices, ensuring that your deployment remains flexible and responsive to changes.

marcschier, 2024-10-06:
Updated documentation where necessary with additional information.

closed 2024-10-06 by marcschier