Azure/Industrial-IoT #2327: Heartbeat duplicate values

b-enigma-con, 2024-08-30:
Our solution requires regular polling of nodes. Even if the data didn't change, we require the last known value periodically. We're monitoring between 500 and 1200 nodes, and we require an interval of 5 seconds. At several sites we noticed that the cyclic read was not performing very well (lots of server queue overflows). I suspect because reading 500-1200 nodes every 5 seconds might be problematic for some OPC UA servers...

I'm currently diving into the heartbeat again, to see if that would fit our needs and if we get a better performance. Whatever configuration I'm applying, I see that we often receive double values. 

Our configuration is as follows (publishednodes.json):
```
[
  {
    "EndpointUrl": "opc.tcp://xxx",
    "UseSecurity": false,
    "OpcNodes": [
      {
        "Id": "ns=3;i=1002",
        "OpcPublishingInterval": 10000,
        "OpcSamplingInterval": 10000,
        "HeartbeatInterval": 10
      }
    ]
  }
]
```
I want to use the timestamp property, but I often see (almost) duplicate values for the timestamp. 

```
[{"NodeId":"ns=3;i=1002","EndpointUrl":"opc.tcp://xxx","ApplicationUri":"xxx","DisplayName":"TestField2","Timestamp":"2024-08-30T10:15:21.710435Z","Value":{"Value":-0.800593,"SourceTimestamp":"2024-08-30T10:15:21Z","ServerTimestamp":"2024-08-30T10:15:21.693Z"},"SequenceNumber":1,"ExtensionFields":{}}]

[{"NodeId":"ns=3;i=1002","EndpointUrl":"opc.tcp://xxx","ApplicationUri":"xxx","DisplayName":"TestField2","Timestamp":"2024-08-30T10:15:31.6704261Z","Value":{"Value":-0.800593,"SourceTimestamp":"2024-08-30T10:15:21Z","ServerTimestamp":"2024-08-30T10:15:21.693Z"},"SequenceNumber":2,"ExtensionFields":{}},
 {"NodeId":"ns=3;i=1002","EndpointUrl":"opc.tcp://xxx","ApplicationUri":"xxx","DisplayName":"TestField2","Timestamp":"2024-08-30T10:15:31.6939288Z","Value":{"Value":1.120031,"SourceTimestamp":"2024-08-30T10:15:31Z","ServerTimestamp":"2024-08-30T10:15:31.691Z"},"SequenceNumber":3,"ExtensionFields":{}}]

[{"NodeId":"ns=3;i=1002","EndpointUrl":"opc.tcp://xxx","ApplicationUri":"xxx","DisplayName":"TestField2","Timestamp":"2024-08-30T10:15:41.6726651Z","Value":{"Value":1.120031,"SourceTimestamp":"2024-08-30T10:15:31Z","ServerTimestamp":"2024-08-30T10:15:31.691Z"},"SequenceNumber":4,"ExtensionFields":{}},
 {"NodeId":"ns=3;i=1002","EndpointUrl":"opc.tcp://xxx","ApplicationUri":"xxx","DisplayName":"TestField2","Timestamp":"2024-08-30T10:15:41.695573Z","Value":{"Value":-0.56948739999999998,"SourceTimestamp":"2024-08-30T10:15:41Z","ServerTimestamp":"2024-08-30T10:15:41.691Z"},"SequenceNumber":5,"ExtensionFields":{}}]

[{"NodeId":"ns=3;i=1002","EndpointUrl":"opc.tcp://xxx","ApplicationUri":"xxx","DisplayName":"TestField2","Timestamp":"2024-08-30T10:15:51.6715363Z","Value":{"Value":-0.56948739999999998,"SourceTimestamp":"2024-08-30T10:15:41Z","ServerTimestamp":"2024-08-30T10:15:41.691Z"},"SequenceNumber":6,"ExtensionFields":{}},
 {"NodeId":"ns=3;i=1002","EndpointUrl":"opc.tcp://xxx","ApplicationUri":"xxx","DisplayName":"TestField2","Timestamp":"2024-08-30T10:15:51.6981644Z","Value":{"Value":-1.181163,"SourceTimestamp":"2024-08-30T10:15:51Z","ServerTimestamp":"2024-08-30T10:15:51.691Z"},"SequenceNumber":7,"ExtensionFields":{}}]

[{"NodeId":"ns=3;i=1002","EndpointUrl":"opc.tcp://xxx","ApplicationUri":"xxx","DisplayName":"TestField2","Timestamp":"2024-08-30T10:16:01.6740468Z","Value":{"Value":-1.181163,"SourceTimestamp":"2024-08-30T10:15:51Z","ServerTimestamp":"2024-08-30T10:15:51.691Z"},"SequenceNumber":8,"ExtensionFields":{}},
 {"NodeId":"ns=3;i=1002","EndpointUrl":"opc.tcp://xxx","ApplicationUri":"xxx","DisplayName":"TestField2","Timestamp":"2024-08-30T10:16:01.6936378Z","Value":{"Value":-0.99987899999999996,"SourceTimestamp":"2024-08-30T10:16:01Z","ServerTimestamp":"2024-08-30T10:16:01.691Z"},"SequenceNumber":9,"ExtensionFields":{}}]
```

What am I missing? What combination for the configuration do I need to always have one single value per node within the configured interval for the heartbeat?

marcschier, 2024-09-01:
Unfortunately, that is the way it works.  The heartbeat sits and fires on a (in this case 10s) timer reset when a new value arrives, on a different thread essentially as the data is received.  So, if you sample every 10 seconds but also heartbeat every 10 seconds, then every 10 seconds you likely get first the heartbeat and millis later the actual value.  If reverse order, no problem.  I guess what you want is to not see the actually received value and use the "PeriodicLKV/PeriodicLKG" behavior.  We could add that new behavior, e.g., calling it "PeriodicLKVDropReceived".  Then you could lower the sampling rate and publishing rate and would get ever n second the LKV or LKG.  Would that be what you are looking for?

marcschier, 2024-09-03:
Please see docs, there are now 2 new behaviors that "drop the actual value" and only send the periodic changes.  Periodic is not periodic per se, it is also timer driven (which has skew based on cpu usage) and the timer is stopped and started on reconnects, so not periodically firing while disconnected.