Azure/Industrial-IoT #2329: EndpointURL in PubSub message -> DataSetFieldContentMask?

b-enigma-con, 2024-08-30:
We require the endpoint URL in an OPC Publisher telemetry message. We see that this field is included in the "Samples" format, but not in the PubSub message (by default). According to this site [Message formats](https://azure.github.io/Industrial-IoT/opc-publisher/messageformats.html) it should be possible (see column "DataSetFieldContentMask"). We can't find the answer in the documentation. Is it possible to configure to include the EndpointURL field in the PubSub message?

marcschier, 2024-09-01:
These flags are there to instruct the old legacy samples encoder to write the endpoint url.  But the pub sub messages do not understand them.  We could add a way to auto populate the extension fields with endpoint url.  Of course until this is the case you could just add any key value as extension field.  Making it a feature request.

marcschier, 2024-09-02:
The way I have implemented it for the JSON PubSub encodings is to add extra fields to the data set (message) if the --fm=True flag is set or "FullNetworkMessage" message mode is chosen instead of "PubSub".  If the message would already contain a EndpointUrl or ApplicationUri field, then the either is not added (to avoid same keys in the json object).  So -- in 2.9.12 you can set --fm=true with PubSub and get the endpoint uri in every message.

closed 2024-09-03 by marcschier

marcschier, 2024-09-05:
mcr.microsoft.com/iotedge/opc-publisher:2.9.12-preview1 is available to test the --fm=true flag with pub/sub messages.