Azure/Industrial-IoT #2334: No endpoint found that matches connection of session

NikolaiFalke, 2024-09-03:
**Bug Report**

**Describe the bug**
Connecting to `opc.tcp://10.x.x.x:48080/OPC.MyGateway_2A4BA307` using OpcPublisher 2.9.11.2 results in "No endpoint found that matches connection of session" error.

**To Reproduce**
1. Configure `azure iot edge` with `OpcPublisher` 2.9.11.2.
2. Set `publishednodes.json` to point to `opc.tcp://10.x.x.x:48080/OPC.MyGateway`.
3. Start OpcPublisher.
4. Observe the error in the logs.

**Expected behavior**
Successful connection to the OPC server endpoint without errors.

**Additional context**
- **Azure IoT Edge Configuration:**
  ```
  "name": "OpcPublisher",
  "image": "mcr.microsoft.com/iotedge/opc-publisher:2.9.11",
  "commandLine": [
    "--aa",
    "--loglevel=Trace",
    "--fetchdisplayname=true",
    "--pf=/appdata/publishednodes.json",
    "--di=60",
    "--an=opcpublisher",
    "--sn=CN=opcpublisher,O=myCompany",
    "--au=urn:localhost:opcpublisher:microsoft",
    "--pki=/pki",
    "--apw=mypw"
  ],
  "mappedFiles": [
    {
      "volume": "iiotedge",
      "inContainer": "/appdata"
    },
    {
      "onHost": "/tmp/opcpki",
      "inContainer": "/pki"
    }
  ]
  ```
- **publishednodes.json:**
  ```
  [
    {
      "EndpointUrl": "opc.tcp://10.x.x.x:48080/OPC.MyGateway",
      "UseSecurity": true,
      "OpcAuthenticationMode": "Anonymous",
      "EndpointSecurityMode": "SignAndEncrypt",
      "EndpointSecurityPolicy": "Basic256Sha256",
      "OpcNodes": [
        {
          "MeasurementName": "InterfaceState",
          "Id": "i=2258",
          "HeartbeatInterval": 3600,
          "OpcSamplingInterval": 1000,
          "OpcPublishingInterval": 1000,
          "HotNode": false,
          "IsNumeric": false
        }
      ]
    }
  ]
  ```

some parts of the logs:

          Starting OpcPublisher module version 2.9.11.2...
[24-09-03 14:33:44.8200] info: Azure.IIoT.OpcUa.Publisher.Services.RuntimeStateReporter[0]
      Generating new Api Key in IoTEdge store...
[24-09-03 14:33:45.3761] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://10.x.x.x:48080/OPC.MyGateway_2A4BA307 [state:Connecting|refs:1]: Discovery endpoint opc.tcp://10.x.x.x:48080/OPC.MyGateway returned endpoints. Selecting endpoint opc.tcp://10.x.x.x:48080/OPC.MyGateway with SecurityMode SignAndEncrypt and Basic256Sha256 SecurityPolicyUri from:
            #006: opc.tcp://10.x.x.x:48080/OPC.MyGateway|Sign [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256]
            #106: opc.tcp://10.x.x.x:48080/OPC.MyGateway|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256]
            #008: opc.tcp://10.x.x.x:48080/OPC.MyGateway|Sign [http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep]
            #108: opc.tcp://10.x.x.x:48080/OPC.MyGateway|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep]
            #010: opc.tcp://10.x.x.x:48080/OPC.MyGateway|Sign [http://opcfoundation.org/UA/SecurityPolicy#Aes256_Sha256_RsaPss]
            #110: opc.tcp://10.x.x.x:48080/OPC.MyGateway|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Aes256_Sha256_RsaPss]
[24-09-03 14:33:45.4247] info: Azure.IIoT.OpcUa.Publisher.Services.RuntimeStateReporter[0]
      Using server certificate with private key from workload API...
[24-09-03 14:33:45.5191] info: Azure.IIoT.OpcUa.Publisher.Services.RuntimeStateReporter[0]
      Stored new Certificate in IoTEdge store (and scheduled renewal after 29.14:33:45).
[24-09-03 14:33:45.5193] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherModule[0]
      OpcPublisher module version 2.9.11.2 started.
[24-09-03 14:33:45.5248] info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
[24-09-03 14:33:45.5252] info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
[24-09-03 14:33:45.5253] info: Microsoft.Hosting.Lifetime[0]
      Content root path: /app
[24-09-03 14:33:45.5262] dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
[24-09-03 14:33:45.5748] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://10.x.x.x:48080/OPC.MyGateway_2A4BA307 [state:Connecting|refs:1]: No endpoint found that matches connection of session opc.tcp://10.x.x.x:48080/OPC.MyGateway_2A4BA307.
[24-09-03 14:33:45.5758] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]

NikolaiFalke, 2024-09-04:
Thank you it got me one step further. Now I receive the following error. Is it possible that something is realy wrong on the server side and the other clients just gracefully accept the connection? Can I also somehow force a connection even when the server side configuration is not optimal?

[24-09-04 06:41:04.6653] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherModule[0]
      Starting OpcPublisher module version 2.9.11.2...
[24-09-04 06:41:04.6953] info: Azure.IIoT.OpcUa.Publisher.Services.RuntimeStateReporter[0]
      Generating new Api Key in IoTEdge store...
[24-09-04 06:41:05.2807] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://10.x.x.x:48080/OPC.MyGateway_44E31B9F [state:Connecting|refs:1]: Discovery endpoint opc.tcp://10.x.x.x:48080/OPC.MyGateway returned endpoints. Selecting endpoint opc.tcp://10.x.x.x:48080/OPC.MyGateway with SecurityMode SignAndEncrypt and http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 SecurityPolicyUri from:
            #006: opc.tcp://10.x.x.x:48080/OPC.MyGateway|Sign [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256]
            #106: opc.tcp://10.x.x.x:48080/OPC.MyGateway|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256]
            #008: opc.tcp://10.x.x.x:48080/OPC.MyGateway|Sign [http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep]
            #108: opc.tcp://10.x.x.x:48080/OPC.MyGateway|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep]
            #010: opc.tcp://10.x.x.x:48080/OPC.MyGateway|Sign [http://opcfoundation.org/UA/SecurityPolicy#Aes256_Sha256_RsaPss]
            #110: opc.tcp://10.x.x.x:48080/OPC.MyGateway|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Aes256_Sha256_RsaPss]
[24-09-04 06:41:05.4021] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://10.x.x.x:48080/OPC.MyGateway_44E31B9F [state:Connecting|refs:1]: Endpoint #106: opc.tcp://10.x.x.x:48080/OPC.MyGateway|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256] selected!
[24-09-04 06:41:05.4321] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #1 - opc.tcp://10.x.x.x:48080/OPC.MyGateway_44E31B9F [state:Connecting|refs:1]: Creating session opc.tcp://10.x.x.x:48080/OPC.MyGateway_44E31B9F with endpoint opc.tcp://10.x.x.x:48080/OPC.MyGateway...
[24-09-04 06:41:05.4398] info: Azure.IIoT.OpcUa.Publisher.Services.RuntimeStateReporter[0]
      Using server certificate with private key from workload API...
[24-09-04 06:41:05.5084] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      Accepting untrusted peer certificate FEF41023F677D6019D60D94ED6E9B1AD0C6C38FA, 'CN=IoT Gateway' due to AutoAccept(UntrustedCertificates) set!
[24-09-04 06:41:05.5415] info: Azure.IIoT.OpcUa.Publisher.Services.RuntimeStateReporter[0]
      Stored new Certificate in IoTEdge store (and scheduled renewal after 29.06:41:05).
[24-09-04 06:41:05.5417] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherModule[0]
      OpcPublisher module version 2.9.11.2 started.
[24-09-04 06:41:05.5471] info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
[24-09-04 06:41:05.5475] info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
[24-09-04 06:41:05.5477] info: Microsoft.Hosting.Lifetime[0]
      Content root path: /app
[24-09-04 06:41:05.5485] dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
[24-09-04 06:41:13.5813] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSession[0]
      : Session disposed.
[24-09-04 06:41:13.5882] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://10.x.x.x:48080/OPC.MyGateway_44E31B9F [state:Connecting|refs:1]: BadNoCommunication 'Error establishing a connection: Error received from remote host: Could not verify security on OpenSecureChannel request.' => Error
[24-09-04 06:41:13.5883] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://10.x.x.x:48080/OPC.MyGateway_44E31B9F [state:Error|refs:1]: Session opc.tcp://10.x.x.x:48080/OPC.MyGateway_44E31B9F with opc.tcp://10.x.x.x:48080/OPC.MyGateway changed from Connecting to Error
[24-09-04 06:41:13.5885] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #2 - opc.tcp://10.x.x.x:48080/OPC.MyGateway_44E31B9F [state:Error|refs:1]: Failed to connect to opc.tcp://10.x.x.x:48080/OPC.MyGateway: Error establishing a connection: Error received from remote host: Could not verify security on OpenSecureChannel request....
[24-09-04 06:41:13.5894] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://10.x.x.x:48080/OPC.MyGateway_44E31B9F [state:Error|refs:1]: Retrying connecting session in 00:00:01...
[24-09-04 06:41:13.5901] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Connect: Event opc.tcp://10.x.x.x:48080/OPC.MyGateway_44E31B9F [state:Error|refs:1] in State Connecting processed.
[24-09-04 06:41:13.5904] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://10.x.x.x:48080/OPC.MyGateway_44E31B9F [state:Error|refs:1]: Processing event SubscriptionManage in State Connecting...
[24-09-04 06:41:13.5911] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]

marcschier, 2024-09-04:
You must configure mutual authentication by adding the publisher "own/cert" certificate to the server trust list (in the server PKI). Tools are server specific, but normally it works through the file system.  

Vice versa, while you have "AutoAccept" (--aa) configured, for production you should add the server cert to the OPC Publisher trust list (see documentation about certificate management).

NikolaiFalke, 2024-09-04:
Thank you for your reply. At first the server certificated was send to rejected. Then we started from scratch and created new certificates on the server side. i put the certificate into trusted/certs like in the documentation where also the others were located. Now the connections seems to work but after some time the publisher is killed and restarts again. We dont see any reason why the certificate may be not ok. Do you have one final idea for us?

```plaintext
[24-09-04 10:57:41.5205] dbug: Furly.Azure.IoT.Edge.Services.IoTEdgeHubSdkClient[0]
      Connecting all clients to myCompany-pokini-3249278s006 using MqttOverTcp.
[24-09-04 10:57:41.6268] info: Furly.Azure.IoT.Edge.Services.IoTEdgeHubSdkClient[0]
      Running in iotedge context.
[24-09-04 10:57:43.1816] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      Own certificate Subject 'CN=opcPublisher, O=myCompany' (Thumbprint: 2A5C262101919748554619D13FDD7268E4A5E615) loaded.
[24-09-04 10:57:43.7668] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      Application own certificate store contains 2 certs.
[24-09-04 10:57:43.7732] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      01: Subject 'CN=opcPublisher, O=myCompany' (Thumbprint: 2A5C262101919748554619D13FDD7268E4A5E615)
[24-09-04 10:57:43.7737] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      02: Subject 'DC=localhost, O=myCompany, CN=opcPublisher' (Thumbprint: 793D927DD29AB9B1E031AEF30EAAA3BB01AA902A)
[24-09-04 10:57:43.7745] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      Trusted issuer store contains 0 certs.
[24-09-04 10:57:43.7759] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      Trusted issuer store has 0 CRLs.
[24-09-04 10:57:43.7777] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      Trusted peer store contains 4 certs.
[24-09-04 10:57:43.7780] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      01: Subject 'CN=IoT Gateway' (Thumbprint: 221C8630740A827B151A841D33C3F135F194A645)
[24-09-04 10:57:43.7781] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      02: Subject 'CN=opcPublisher, O=myCompany' (Thumbprint: 2A5C262101919748554619D13FDD7268E4A5E615)
[24-09-04 10:57:43.7782] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      03: Subject 'DC=localhost, O=myCompany, CN=opcPublisher' (Thumbprint: 793D927DD29AB9B1E031AEF30EAAA3BB01AA902A)
[24-09-04 10:57:43.7783] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      04: Subject 'CN=IoT Gateway' (Thumbprint: FEF41023F677D6019D60D94ED6E9B1AD0C6C38FA)
[24-09-04 10:57:43.7785] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      Trusted peer store has 0 CRLs.
[24-09-04 10:57:43.7787] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      Rejected certificate store contains 0 certs.
[24-09-04 10:57:43.8612] dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
[24-09-04 10:57:43.9661] warn: Microsoft.AspNetCore.DataProtection.Repositories.FileSystemXmlRepository[60]
      Storing keys in a directory '/root/.aspnet/DataProtection-Keys' that may not be persisted outside of the container. Protected data will be unavailable when container is destroyed. For more information go to https://aka.ms/aspnet/dataprotectionwarning
[24-09-04 10:57:43.9703] info: Microsoft.AspNetCore.DataProtection.KeyManagement.XmlKeyManager[62]
      User profile is available. Using '/root/.aspnet/DataProtection-Keys' as key repository; keys will not be encrypted at rest.
[24-09-04 10:57:44.0444] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherModule[0]
      0: Module OpcPublisher closed due to Communication_Error.
[24-09-04 10:57:44.0802] fail: Furly.Azure.IoT.Edge.Services.IoTEdgeTwinClient[0]
      Failed to process changes, existing.
      System.AggregateException: One or more errors occurred. (TLS authentication error.)
       ---> System.Security.Authentication.AuthenticationException: TLS authentication error.
       ---> System.AggregateException: One or more errors occurred. (The remote certificate is invalid because of errors in the certificate chain: PartialChain)
       ---> System.Security.Authentication.AuthenticationException: The remote certificate is invalid because of errors in the certificate chain: PartialChain
         at System.Net.Security.SslStream.SendAuthResetSignal(ReadOnlySpan`1 alert, ExceptionDispatchInfo exception)
         at System.Net.Security.SslStream.CompleteHandshake(SslAuthenticationOptions sslAuthenticationOptions)
         at System.Net.Security.SslStream.ForceAuthenticationAsync[TIOAdapter](Boolean receiveFirst, Byte[] reAuthenticationData, CancellationToken cancellationToken)
         --- End of inner exception stack trace ---
         at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenInternalAsync(CancellationToken cancellationToken)
         at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsync(CancellationToken cancellationToken)
         at Microsoft.Azure.Devices.Client.Transport.ProtocolRoutingDelegatingHandler.OpenAsync(CancellationToken cancellationToken)
         at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.<>c__DisplayClass27_0.<<ExecuteWithErrorHandlingAsync>b__0>d.MoveNext()
      --- End of stack trace from previous location ---
[... cut ...]
         --- End of inner exception stack trace ---
         at Furly.Azure.IoT.Edge.Services.IoTEdgeHubSdkClient.CreateAdapterAsync(IotHubConnectionStringBuilder cs, List`1 settings)
         at Furly.Azure.IoT.Edge.Services.IoTEdgeHubSdkClient.SetMethodHandlerAsync(MethodCallback methodHandler, Object userContext, CancellationToken ct)
         at Furly.Azure.IoT.Edge.Services.IoTEdgeRpcServer.ConnectAsync(IRpcHandler server, CancellationToken ct)
[24-09-04 10:57:46.4239] info: Azure.IIoT.OpcUa.Publisher.Discovery.NetworkDiscovery[0]
```

NikolaiFalke, 2024-09-04:
if this helps for more context. This is the Log from the server side
[image]

marcschier, 2024-09-05:
@NikolaiFalke, if you hit issues with certificates typically it is because the OPC Publisher certificate is not persisted, and therefore always generated when the OPC Publisher is restarted.  This 

"but after some time the publisher is killed and restarts again" 

seems to indicate what you are facing is due to it.  The newly generated certificate is then again not trusted by the server.  

To persist the application certificate and re-use it across restarts, mount the pki folder to the host file system using a docker volume mount.  That is described in the docs, too, I believe.

marcschier, 2024-09-05:
The docker mount configuration does not look right to me (it should be "binds", no?  This is how it should look like in IoT Edge deployment manifest: https://github.com/Azure/Industrial-IoT/blob/7e528daaf8022a5664c4a353035d1cb350799c12/deploy/iotedge/eflow-setup.json#L44.  

Presuming you use IoT Edge as per the error above.  On that front - the chain issue looks like an IoT Edge issue to me, can you follow up on the iot edge side with support, as I think the TLS certificate of the IoT Edge installation is not correct.  https://github.com/Azure/Industrial-IoT/blob/main/deploy/iotedge/eflow-setup.ps1 creates you a eflow iot edge with publisher running (or without), and that is a breeze to get up and running.

marcschier, 2024-09-05:
#2337

closed 2024-09-06 by marcschier