Azure/Industrial-IoT #2335: Heartbeat Issue

e-augusto, 2024-09-03:
Sometimes a day the publisher publishes null data for no real reason. It seems to me like a heartbeat error, as if it "lost" its time. And it happens with tags that keep the same value for a long time, such as a machine running status. Honestly, I think we should just ignore these records, but I'm being pressured to fix these holes.  Do you have any suggestions?

Thanks!

[image]
[image]
[image]

marcschier, 2024-09-04:
From which version was this publisher forked?  I see a version 7.0.2 which is unknown.

e-augusto, 2024-09-04:
I don't understand much about installing opc Publisher, but when I use the list command in Linux, this is the version that appears.

[image]

marcschier, 2024-09-05:
@e-augusto, would love to help, but I dont know anything about this version / product - this is not a OPC Publisher supported via this repository/github project, please follow up with the person/company/developer you or your company obtained it from to have them address the issue you are facing.  

The latest version of OPC Publisher in this repo is mcr.microsoft.com/iotedge/opc-publisher:2.9.11.  Which is also the only version supported via support path ([see here](readme.md)).

closed 2024-09-05 by marcschier