Azure/Industrial-IoT #2342: industrial-iot-services failing after warmup

dplumm2, 2024-09-11:
**Describe the bug**
Container <application-name> for site <site-name> has exited, failing site start

**To Reproduce**
Followed the .\deploy,
created new RG and all resources
No errors
AAD working and setup in Entra ID

**Expected behavior**
Web app on site to startup

**Screenshots**
[image]

**Desktop (please complete the following information):**
Edge and azure portal

**Additional context**
Here is a snip of my logstream

2024-09-11T17:10:47.5622364Z Configure Services : 05.10.47.562048
2024-09-11T17:10:49.4941414Z Configure : 05.10.49.487388
2024-09-11T17:10:50.1496929Z Setting Up Routes : 05.10.50.149534
2024-09-11T17:10:50.7286659Z Exiting Configure : 05.10.50.728508
2024-09-11T17:10:50.8381688Z [40m[1m[33mwarn[39m[22m[49m: Microsoft.AspNetCore.DataProtection.KeyManagement.XmlKeyManager[35]
2024-09-11T17:10:50.8382549Z       No XML encryptor configured. Key {a6cf9851-2ee5-47eb-b90b-6e93f09ea960} may be persisted to storage in unencrypted form.
2024-09-11T17:10:50.9753330Z Hosting environment: Production
2024-09-11T17:10:50.9770314Z Content root path: /opt/Kudu
2024-09-11T17:10:50.9782744Z Now listening on: http://0.0.0.0:8181
2024-09-11T17:10:50.9959394Z Application started. Press Ctrl+C to shut down./home/LogFiles/2024_09_11_lw1sdlwk0005WH_docker.log  (https://xxxxxxxxxxx.scm.azurewebsites.net/api/vfs/LogFiles/2024_09_11_lw1sdlwk0005WH_docker.log)
2024-09-11T21:06:25.421Z INFO  - Pull Image successful, Time taken: 0 Seconds
2024-09-11T21:06:26.447Z INFO  - Starting container for site
2024-09-11T21:06:26.447Z INFO  - docker run -d --expose=9080 --name xxxxxxxxxxxt_0_41fedca0 -e WEBSITE_USE_DIAGNOSTIC_SERVER=false -e WEBSITES_PORT=9080 -e WEBSITE_SITE_NAME=xxxxxxxxxxx -e WEBSITE_AUTH_ENABLED=False -e WEBSITE_ROLE_INSTANCE_ID=0 -e WEBSITE_HOSTNAME=xxxxxxxxxxx.azurewebsites.net -e WEBSITE_INSTANCE_ID=0ad77132a4a61ed67b078d3a655d40a58d4b822256e552ee0bac37f77112e58b -e HTTP_LOGGING_ENABLED=1 mcr.microsoft.com/iot/industrial-iot-services-all:latest
2024-09-11T21:06:29.352Z INFO  - Initiating warmup request to container xxxxxxxxxxx_0_41fedca0_msiProxy for site xxxxxxxxxxx
2024-09-11T21:06:29.374Z INFO  - Container xxxxxxxxxxxt_0_41fedca0_msiProxy for site xxxxxxxxxxx initialized successfully and is ready to serve requests.
2024-09-11T21:06:29.381Z INFO  - Initiating warmup request to container xxxxxxxxxxxt_0_41fedca0 for site xxxxxxxxxxx
2024-09-11T21:06:47.928Z INFO  - Waiting for response to warmup request for container xxxxxxxxxxx_0_41fedca0. Elapsed time = 18.5750389 sec
2024-09-11T21:06:47.977Z ERROR - Container xxxxxxxxxxxt_0_41fedca0 for site xxxxxxxxxxx has exited, failing site start
2024-09-11T21:06:47.997Z ERROR - Container xxxxxxxxxxx_0_41fedca0 didn't respond to HTTP pings on port: 9080, failing site start. See container logs for debugging.
2024-09-11T21:06:49.347Z INFO  - Stopping site xxxxxxxxxxx because it failed during startup./home/LogFiles/2024_09_11_lw1sdlwk0005WH_msi_docker.log  (https://xxxxxxxxxxx.scm.azurewebsites.net/api/vfs/LogFiles/2024_09_11_lw1sdlwk0005WH_msi_docker.log)Ending Log Tail of existing logs ---Starting Live Log Stream ---

I read in another post trying change from latest in deploy.ps1 to 2.9.0-preview4 version, but that was over 1 year old and was a slightly different issue.

Please advise what I'm doing wrong.  I note that none of my environment variable are configured except DOCKER_, WEBSITES_PORT and AZURE_, none of the PCS_ variables are configured, i did try to add them but the container still didn't start.

Many thanks

marcschier, 2024-09-11:
Did you use the deploy script?  Were there any deployment errors after the script completed?  Can you set the version arg of the script to "2.9.11"

dplumm2, 2024-09-12:
I did use the script and it all seemed to work ok:

Here is some of the output of the terminal:

Start time: 17:28
Using released latest images from mcr.microsoft.com.
Using Standard_B2ms as VM size for Linux IoT Edge gateway simulations...
Using Standard_B1ms as VM size for all OPC PLC simulation host machines...

Registering client and services AAD applications in your tenant...
Using passed context (Account xxxxxxxxxxxxx, Tenant xxxxxxxxxxxxxxxx3)
WARNING: Upcoming breaking changes in the cmdlet 'Get-AzAccessToken' :
The Token property of the output type will be changed from String to SecureString. Add the [-AsSecureString] switch to avoid the impact of this upcoming breaking change.
- The change is expected to take effect in Az version : '13.0.0'
- The change is expected to take effect in Az.Accounts version : '4.0.0'
Note : Go to https://aka.ms/azps-changewarnings for steps to suppress this breaking change warning, and other information on breaking changes in Azure PowerShell.
Updating client application 'strongfordat-client' in Tenant 'xxxxxxxxxxxxx'.
  Added redirect uris.
Updating web client application 'xxxxxxx-web' in Tenant 'xxxxxxxxxxxxxx'.
  Added redirect uris.
  Added a secret.
Updating service application 'xxxxxxxxx-service' in Tenant 'metroplummer.com'.
  Added identifiers.
  Added a secret.
  Added optional claims.
  Added known applications.
  Added pre-authorized applications.
  Assigned Reader role to Davidxxxxxxxr.com.
  Assigned Writer role to Davidxxxxxxxr.com.
  Assigned Admin role to Davidxxxxxxxr.com.
Client application 'xxxxxxxxxxxx-client' updated with required resource access.
  Admin consent granted to native client application.
Web application 'xxxxxxxxxxx-web' updated with required resource access.
  Admin consent granted to web application.
Client and services AAD applications registered...

Starting deployment 'latest'...
Deployment 'latest' succeeded.

I've had maybe 6 attempts at deploying, the Hub and VMs always deploy fine and work, I can set Public Ip and update NSGs no problems for SSH and RDP and so on, but the web app never seems to start.

I will set to 2.9.11 and revert to you.

Thanks

dplumm2, 2024-09-12:
redeployed using 2.9.11 and it seemed to be running, I tried to access the site and it subsequently failed.

2024-09-12T13:07:33.100Z ERROR - Container strongfordat_0_7ba9d60e for site strongfordat has exited, failing site start
2024-09-12T13:07:33.301Z ERROR - Container strongfordat_0_7ba9d60e didn't respond to HTTP pings on port: 9080, failing site start. See container logs for debugging.
2024-09-12T13:07:33.431Z INFO  - Stopping site strongfordat because it failed during startup./home/LogFiles/2024_09_12_lw0sdlwk000655_msi_docker.log  (https://strongfordat.scm.azurewebsites.net/api/vfs/LogFiles/2024_09_12_lw0sdlwk000655_msi_docker.log)Ending Log Tail of existing logs ---Starting Live Log Stream ---

marcschier, 2024-09-13:
Would you be able to get the container logs themselves?  I would like to understand the reason why the container exits.

marcschier, 2024-10-03:
@dplumm2, please can you get the part of the container logs that show the container exiting?

marcschier, 2024-10-06:
Closing due to inactivity.

closed 2024-10-06 by marcschier