Azure/Industrial-IoT #2343: Call OPC UA method via REST Api and ExtensionObject as input argument

Haeavar, 2024-09-16:
**Describe the bug**
Im not sure whether its a bug or missing feature. 
We try to call a method with complex input argument.

`"inputArguments": [
      {
        "name": "JobOrder",
        "description": "Contains information defining the job order with all parameters and any material, equipment, or physical asset requirements associated with the order.",
        "type": {
          "nodeClass": "dataType",
          "displayName": "ISA95JobOrderDataType",
          "nodeId": "nsu=http://opcfoundation.org/UA/ISA95-JOBCONTROL_V2;i=3008",
          "description": "Defines the information needed to schedule and execute a job.",
          "browseName": "nsu=http://opcfoundation.org/UA/ISA95-JOBCONTROL_V2;ISA95JobOrderDataType",
          "value": null,
          "sourcePicoseconds": 0,
          "sourceTimestamp": "0001-01-01T00:00:00",
          "serverPicoseconds": 0,
          "serverTimestamp": "2024-09-12T10:31:29.9620476Z",
          "writeMask": 0,
          "userWriteMask": 0,
          "isAbstract": false,
          "dataTypeDefinition": {
            "TypeId": "i=15066",
            "Encoding": "Json",
            "Body": {
              "DefaultEncodingId": "http://opcfoundation.org/UA/ISA95-JOBCONTROL_V2#i=5014",
              "BaseDataType": "ExtensionObject",
              "StructureType": 1,
              "Fields": [
                {
                  "Name": "JobOrderID",
                  "Description": {
                    "Text": "An identification of the Job Order."
                  },
                  "DataType": "String",
                  "ValueRank": -1,
                  "ArrayDimensions": [],
                  "MaxStringLength": 0,
                  "IsOptional": false
                },
                .....
              ]
            }
          },
          "children": false
        },
        "defaultValue": null,
        "arrayDimensions": []
      }
    ],`

We call the method with following body on URL "/v2/call"
`{
  "connection": {
    "endpoint": {
      "url": "opc.tcp://host.opcsrv.internal:4842",
      "securityMode": "None"
    }
  },
  "request": {
    "objectId": "nsu=http://trumpf.com/UA/BendIsa95/;i=5002",
    "methodId": "nsu=http://trumpf.com/UA/BendIsa95/;i=7000",
    "arguments": [
      {
        "dataType": "nsu=http://opcfoundation.org/UA/ISA95-JOBCONTROL_V2;i=3008",
        "value": {
          "JobOrderId": "Order1"
        }
      }
    ]
  }
}`

We get following response:
`{
  "results": [],
  "errorInfo": {
    "statusCode": 2158690304,
    "symbolicId": "BadInvalidArgument"
  }
}`

Do we use a wrong value or are complex types currently not supported? Unfortunately, the opcplc sample server does not have any method with complex input arguments, so I cant reproduce with it.

You you give us a hint? 

Thanks in advance!

**Additional context**
OpcPublisher 2.9.11.2+63bedd6b75 (.NET 8.0.8/linux-x64/OPC Stack 1.5.374.78)

Haeavar, 2024-09-18:
It is working if all type information are passed correctly. Maybe you could add a sample to your documentation.
And im wondering, if these information can be used as default? The type is already defined in the `dataType` field and the default encoding could be always json. This would make calls a lot easier.

`
"arguments": [
      {
        "dataType": "nsu=http://opcfoundation.org/UA/ISA95-JOBCONTROL_V2;i=3008",
        "value": {
            "TypeId": "nsu=http://opcfoundation.org/UA/ISA95-JOBCONTROL_V2;i=3008",
            "Encoding": "json",
            "Body": { "JobOrderId": "workingTest" },
            "Type": 22
        }
      }
    ]
`

The issue can be closed...

marcschier, 2024-09-18:
Super, yes I would have pointed to https://reference.opcfoundation.org/Core/Part6/v105/docs/5.4.2.16 but there are indeed some optimizations in the json decoder that should kick in, and allow the first to work also, I will keep open to take another look.

closed 2024-10-04 by marcschier