Azure/Industrial-IoT #2344: OPC Publisher v. 2.9.11: Kepware reinitialize results in "Unexpected error sending publish request".

b-enigma-con, 2024-09-18:
**Describe the bug**
OPC Publisher v. 2.9.11 is running with no issues for days. The underlying OPC UA server is KepServerEX, which has a reinitialize function. This function is used when the runtime server needs a reset disconnects all clients. The OPC Publisher does not seem to be able to successfully re-establish a connection afterwards.

**To Reproduce**
Steps to reproduce the behavior:
1. Run and configure OPC Publisher v. 2.9.11 with KepServerEX OPC UA server
2. Reinitialize KepServerEX
3. Inspect OPC Publisher logs

**Expected behavior**
I would expect the OPC Publisher to re-establish the connection.

**Screenshots**
N.A.

**Additional context**
Logging snippet:

[24-09-18 13:19:10.5996] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Unexpected error sending publish request.
      Opc.Ua.ServiceResultException: BadNotConnected
         at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.QueueConnectOperation(WriteOperation operation, Int32 timeout, IServiceRequest request)
         at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.BeginSendRequest(IServiceRequest request, Int32 timeout, AsyncCallback callback, Object state)
         at Opc.Ua.Client.Session.BeginPublish(Int32 timeout)
[24-09-18 13:19:10.6003] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #347060, Reconnecting=False, Error: BadSecureChannelClosed
[24-09-18 13:19:15.5150] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Could not reconnect the Session. System.**NullReferenceException**: Object reference not set to an instance of an object.
         at Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaMonitoredItem.CyclicRead.EnsureSamplerRunning() in /__w/1/s/Industrial-IoT/src/Azure.IIoT.OpcUa.Publisher/src/Stack/Services/OpcUaMonitoredItem.CyclicRead.cs:line 173
         at Opc.Ua.Client.Subscription..ctor(Subscription template, Boolean copyEventHandlers)
         at Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription..ctor(OpcUaSubscription subscription, Boolean copyEventHandlers) in /__w/1/s/Industrial-IoT/src/Azure.IIoT.OpcUa.Publisher/src/Stack/Services/OpcUaSubscription.cs:line 123
         at Opc.Ua.Client.Session..ctor(ITransportChannel channel, Session template, Boolean copyEventHandlers)
         at Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSession..ctor(OpcUaSession session, ITransportChannel channel, Session template, Boolean copyEventHandlers) in /__w/1/s/Industrial-IoT/src/Azure.IIoT.OpcUa.Publisher/src/Stack/Services/OpcUaSession.cs:line 141
         at Opc.Ua.Client.Session.RecreateAsync(Session sessionTemplate, CancellationToken ct)
         at Opc.Ua.Client.Session.RecreateAsync(Session sessionTemplate, ITransportChannel transportChannel, CancellationToken ct)
         at Opc.Ua.Client.DefaultSessionFactory.RecreateAsync(ISession sessionTemplate, ITransportChannel transportChannel, CancellationToken ct)
         at Opc.Ua.Client.SessionReconnectHandler.DoReconnectAsync()
[24-09-18 13:19:20.9716] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Unexpected error sending publish request.
      Opc.Ua.ServiceResultException: BadNotConnected
         at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.QueueConnectOperation(WriteOperation operation, Int32 timeout, IServiceRequest request)
         at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.BeginSendRequest(IServiceRequest request, Int32 timeout, AsyncCallback callback, Object state)
         at Opc.Ua.Client.Session.BeginPublish(Int32 timeout)
[24-09-18 13:19:20.9723] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #347061, Reconnecting=False, Error: BadSecureChannelClosed
[24-09-18 13:19:21.2073] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Unexpected error sending publish request.
      Opc.Ua.ServiceResultException: BadNotConnected
         at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.QueueConnectOperation(WriteOperation operation, Int32 timeout, IServiceRequest request)
         at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.BeginSendRequest(IServiceRequest request, Int32 timeout, AsyncCallback callback, Object state)
         at Opc.Ua.Client.Session.BeginPublish(Int32 timeout)
[24-09-18 13:19:21.2081] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #347062, Reconnecting=False, Error: BadSecureChannelClosed
[24-09-18 13:19:21.3231] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Unexpected error sending publish request.
      Opc.Ua.ServiceResultException: BadNotConnected
         at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.QueueConnectOperation(WriteOperation operation, Int32 timeout, IServiceRequest request)
         at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.BeginSendRequest(IServiceRequest request, Int32 timeout, AsyncCallback callback, Object state)
         at Opc.Ua.Client.Session.BeginPublish(Int32 timeout)

MattCosturos, 2024-09-18:
I too am using KepServer, however I'm not even able to connect. Created issue #2345 to track.

marcschier, 2024-09-21:
Assumption is that Subscription property is null or invalid when  https://github.com/Azure/Industrial-IoT/blob/e192948c2ebced907ad5fa1cbe3178d6e7024bee/src/Azure.IIoT.OpcUa.Publisher/src/Stack/Services/OpcUaMonitoredItem.CyclicRead.cs#L71 is called.  We need some reconnect testing for cyclic reads for sure.

marcschier, 2024-10-06:
Fixed in 2.9.12

closed 2024-10-06 by marcschier