Azure/Industrial-IoT #2345: Updating from 2.9.6 to 2.9.11 results in OPC Publisher unable to connect to kepserver opc server

MattCosturos, 2024-09-18:
We have been using 2.9.6 of OPC Publisher.
Trying to update to use 2.9.11

The 2.9.11 version is unable to connect to our server

Steps to reproduce the behavior:
Update deployment to use 2.9.11 instead of 2.9.6

**Expected behavior**
Based on the minor version changes, I expect my existing configuration to work.  I expect all the existing config options and direct methods to work the same.

Logs here.  The logs seem to indicate the opc publisher module knows to use security (NotNone) but the Sender Cert is none. Is this the problem?

```

 ██████╗ ██████╗  ██████╗    ██████╗ ██╗   ██╗██████╗ ██╗     ██╗███████╗██╗  ██╗███████╗██████╗
██╔═══██╗██╔══██╗██╔════╝    ██╔══██╗██║   ██║██╔══██╗██║     ██║██╔════╝██║  ██║██╔════╝██╔══██╗
██║   ██║██████╔╝██║         ██████╔╝██║   ██║██████╔╝██║     ██║███████╗███████║█████╗  ██████╔╝
██║   ██║██╔═══╝ ██║         ██╔═══╝ ██║   ██║██╔══██╗██║     ██║╚════██║██╔══██║██╔══╝  ██╔══██╗
╚██████╔╝██║     ╚██████╗    ██║     ╚██████╔╝██████╔╝███████╗██║███████║██║  ██║███████╗██║  ██║
 ╚═════╝ ╚═╝      ╚═════╝    ╚═╝      ╚═════╝ ╚═════╝ ╚══════╝╚═╝╚══════╝╚═╝  ╚═╝╚══════╝╚═╝  ╚═╝
                                  2.9.11.2+63bedd6b75 (.NET 8.0.8/linux-x64/OPC Stack 1.5.374.78)

[24-09-18 20:37:28.0516] dbug: Furly.Azure.IoT.Edge.Services.IoTEdgeHubSdkClient[0]
      Connecting all clients to VM_HOST_NAME using AmqpOverTcp.
[24-09-18 20:37:28.0950] info: Furly.Azure.IoT.Edge.Services.IoTEdgeHubSdkClient[0]
      Running in iotedge context.
[24-09-18 20:37:28.6833] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[512]
      Imported the PFX private key for [032FA5E1751BCB839601370A8EB925744FFE2F72].
[24-09-18 20:37:28.7028] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      Own certificate Subject 'CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=VM_HOST_NAME' (Thumbprint: 032FA5E1751BCB839601370A8EB925744FFE2F72) loaded.
[24-09-18 20:37:28.7048] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Checking application instance certificate.
[24-09-18 20:37:28.7393] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[512]
      Imported the PFX private key for [032FA5E1751BCB839601370A8EB925744FFE2F72].
[24-09-18 20:37:28.7406] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Check certificate: [CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=VM_HOST_NAME] [032FA5E1751BCB839601370A8EB925744FFE2F72]
[24-09-18 20:37:28.7424] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Check application instance certificate. [CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=VM_HOST_NAME] [032FA5E1751BCB839601370A8EB925744FFE2F72]
[24-09-18 20:37:28.9586] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Using the ApplicationUri: urn:VM_HOST_NAME:Microsoft.Azure.IIoT:microsoft:
[24-09-18 20:37:28.9647] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      Application own certificate store contains 2 certs.
[24-09-18 20:37:28.9673] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      01: Subject 'CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=9150:da7c:87d7:e037:df54:4e95:b6b3:c1d4' (Thumbprint: 903C1D0C90EB1D762AF91CC45D5B7DAEEEB07695)
[24-09-18 20:37:28.9674] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      02: Subject 'CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=VM_HOST_NAME' (Thumbprint: 032FA5E1751BCB839601370A8EB925744FFE2F72)
[24-09-18 20:37:28.9677] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      Trusted issuer store contains 0 certs.
[24-09-18 20:37:28.9682] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      Trusted issuer store has 0 CRLs.
[24-09-18 20:37:28.9690] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      Trusted peer store contains 3 certs.
[24-09-18 20:37:28.9692] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      01: Subject 'CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=9150:da7c:87d7:e037:df54:4e95:b6b3:c1d4' (Thumbprint: 903C1D0C90EB1D762AF91CC45D5B7DAEEEB07695)
[24-09-18 20:37:28.9693] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      02: Subject 'CN=KEPServerEX/UA Server, O=Unknown, C=US, DC=USEA1SVKEP01.corp.mooginc.com' (Thumbprint: A8D8276CB251465002F2E30E2636BD942084BBA7)
[24-09-18 20:37:28.9694] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      03: Subject 'CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=VM_HOST_NAME' (Thumbprint: 032FA5E1751BCB839601370A8EB925744FFE2F72)
[24-09-18 20:37:28.9694] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      Trusted peer store has 0 CRLs.
[24-09-18 20:37:28.9696] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaApplication[0]
      Rejected certificate store contains 0 certs.
[24-09-18 20:37:28.9955] dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
[24-09-18 20:37:29.0280] warn: Microsoft.AspNetCore.DataProtection.Repositories.FileSystemXmlRepository[60]
      Storing keys in a directory '/root/.aspnet/DataProtection-Keys' that may not be persisted outside of the container. Protected data will be unavailable when container is destroyed. For more information go to https://aka.ms/aspnet/dataprotectionwarning
[24-09-18 20:37:29.0296] info: Microsoft.AspNetCore.DataProtection.KeyManagement.XmlKeyManager[62]
      User profile is available. Using '/root/.aspnet/DataProtection-Keys' as key repository; keys will not be encrypted at rest.
[24-09-18 20:37:29.1822] dbug: Microsoft.AspNetCore.DataProtection.Repositories.FileSystemXmlRepository[37]
[... 144 lines cut ...]
            #051: opc.tcp://myServer:myPort/|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256]
[24-09-18 20:37:31.6769] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://myServer:myPort_457DFB59_myGroup [state:Connecting|refs:1]: Endpoint #051: opc.tcp://myServer:myPort/|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256] selected!
[24-09-18 20:37:32.1737] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 1096434167: CLIENTCHANNEL SOCKET CLOSED: 0346AEFF
[24-09-18 20:37:32.1738] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #1 - opc.tcp://myServer:myPort_457DFB59_myGroup [state:Connecting|refs:1]: Creating session opc.tcp://myServer:myPort_457DFB59_myGroup with endpoint opc.tcp://myServer:myPort/...
[24-09-18 20:37:32.1820] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 0: Token #0 created. CreatedAt=20:37:32.181. Lifetime=3600000.
[24-09-18 20:37:32.1862] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 0: CLIENTCHANNEL SOCKET CLOSED: 030E9943
[24-09-18 20:37:32.1866] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSession[0]
      : Session disposed.
[24-09-18 20:37:32.1870] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://myServer:myPort_457DFB59_myGroup [state:Connecting|refs:1]: BadNoCommunication 'Error establishing a connection: Error received from remote host: An error occurred verifying security.' => Error
[24-09-18 20:37:32.1871] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://myServer:myPort_457DFB59_myGroup [state:Error|refs:1]: Session opc.tcp://myServer:myPort_457DFB59_myGroup with opc.tcp://myServer:myPort changed from Connecting to Error
[24-09-18 20:37:32.1872] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #2 - opc.tcp://myServer:myPort_457DFB59_myGroup [state:Error|refs:1]: Failed to connect to opc.tcp://myServer:myPort/: Error establishing a connection: Error received from remote host: An error occurred verifying security....
[24-09-18 20:37:32.1873] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      opc.tcp://myServer:myPort_457DFB59_myGroup [state:Error|refs:1]: Retrying connecting session in 00:00:01.0320000...
[24-09-18 20:37:32.1874] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      ConnectRetry: Event opc.tcp://myServer:myPort_457DFB59_myGroup [state:Error|refs:1] in State Connecting processed.

```

marcschier, 2024-09-20:
It looks like the server did not trust the OPC publisher certificate, is it in it's trust list after reinit?

marcschier, 2024-09-24:
We had some problems with the format of the cert not being backwards compatible to 2.8 in 2.9 and we corrected that after 2.9.7(?) but unfortunately that shoots down upgrades from previous versions.  To provide a. Workaround we introduced a switch to read the first valid cert in the own folder and use its info to configure the publisher (--cfa).  More info can be found in the readme.md pki section.

closed 2024-09-24 by marcschier

MattCosturos, 2024-09-24:
Thanks for the help. Shame on me for not checking all the version update notes for all the versions I jumped.