Azure/Industrial-IoT #2365: Increasing Memory Consumption

steffenbeermann, 2024-11-12:
**Describe the bug**
We noticed that we are running out of memory sometimes on the devices running IoT Edge with the OPC Publisher. After investigating, we noticed that the OPC Publisher module takes more and more memory over time, see:
[image]

Restarting the module will reset the memory usage (see dip at start of the graph). 
We noticed the same behavior on OPC Publisher 2.9.11 and 2.9.0

**To Reproduce**
We have the following config:
```
{
  "Hostname": "publisher_axh_active",
  "Cmd": [
    "--cl=5",
    "--cf",
    "--aa",
    "--pf=/mount/publishednodes.json",
    "--PkiRootPath=/mount/pki",
    "--si=90",
    "--di=3600"
  ],
  "HostConfig": {
    "Binds": [
      "/home/edge/OPCPublisherMount_Active:/mount"
    ]
  }
}
```

and we have around 20 endpoints configured with a total of around 200.000 Nodes in multiple subscription.

**Expected behavior**
The OPC Publisher does its garbage collection it self and does not built up high memory usages

marcschier, 2024-11-15:
Could you run without Prometheus by setting `--em=False`?  

Also, how does the oom show for you?

steffenbeermann, 2024-11-15:
We rely on the metrics of the OPC publisher module. With disabling Prometheus, we won't get them anymore? But I will try on our test system.

We noticed the issue because the IoT Edge runtime crashed and restarted all modules. It failed for a few hours until it went up again. I will add a screenshot of the memory drop.

marcschier, 2024-11-22:
Could you take a look at the diagnostic log and check the numbers of the data flow pipeline (encoder related, send queue, egress, etc.), check if they are low or increasing?  That could mean we need to tune the data flow path to not hold on to incoming data it cannot send out.

My colleague also found a resource leak in the security handshake of the OPC UA stack.  But it is likely not enough to explain the increases.  Especially the huge increase at the end of the run from 2->4GB.

steffenbeermann, 2024-12-02:
I have this diagnostic info for one endpoint:
```
    "sentMessagesPerSec": 0.12591434486187222,
    "ingestionDuration": "6.17:47:16.9692019",
    "ingressDataChanges": 3494295,
    "ingressValueChanges": 12381772,
    "ingressBatchBlockBufferSize": 48,
    "encodingBlockInputSize": 0,
    "encodingBlockOutputSize": 0,
    "encoderNotificationsProcessed": 3495567,
    "encoderNotificationsDropped": 0,
    "encoderIoTMessagesProcessed": 73343,
    "encoderAvgNotificationsMessage": 47.660540201519055,
    "encoderAvgIoTMessageBodySize": 37767.330201928016,
    "encoderAvgIoTChunkUsage": 9.220539600080082,
    "estimatedIoTChunksPerDay": 0,
    "outgressInputBufferCount": 0,
    "outgressInputBufferDropped": 0,
    "outgressIoTMessageCount": 73336,
    "connectionRetries": 0,
    "opcEndpointConnected": true,
    "monitoredOpcNodesSucceededCount": 6862,
    "monitoredOpcNodesFailedCount": 0,
    "ingressEventNotifications": 0,
    "ingressEvents": 0,
    "encoderMaxMessageSplitRatio": 0,
    "ingressDataChangesInLastMinute": 360,
    "ingressValueChangesInLastMinute": 1144,
    "ingressHeartbeats": 0,
    "ingressCyclicReads": 0
```

I don't exactly understand what you mean with data flow pipeline?

steffenbeermann, 2024-12-02:
> Could you run without Prometheus by setting `--em=False`?
> 
> Also, how does the oom show for you?

I tried with the setting `--em=False` but it seems it did not solved the problem as there is still an slow increase over the last two weeks:
[image]

dimtix, 2025-05-26:
I reproduced the issue locally using Visual Studio with a setup that mirrors our production environment.  
The Publisher was launched from the latest **main** branch with the Visual Studio memory profiler with the following start parameters:

`
"commandLineArgs": "--messagingmode=FullNetworkMessages --BatchSize=10 --LingerTimeoutSeconds=60 --DisableDataSetMetaData=false --ApplicationUri=urn:opc-publisher:machine --BatchTriggerInterval=1000 --UseStandardsCompliantEncoding=True --RootTopicTemplate=opc-publisher --SiteId=machine --DefaultMetaDataUpdateTime=0 --CreatePublishFileIfNotExist=true --DefaultMessageTimeToLive=648000000 --LogLevel=Information --DefaultQueueSize=100 --DisableSessionPerWriterGroup=true --MaxPublishRequests=20 --MqttClientId=publisher --DefaultTransport=Mqtt --DefaultSamplingInterval=50 --PublisherId=machine --TelemetryTopicTemplate={RootTopic}/default/messages/{DataSetWriterGroup} --PublishedNodesFile=c:/temp/pn.json --DiagnosticsTopicTemplate={RootTopic}/default/diagnostics/{DataSetWriterGroup} --EnableRoutingInfo=True --MqttClientConnectionString=HostName=localhost;Port=1883;Protocol='v5' --UnsecureHttpServerPort=8080 --EventsTopicTemplate={RootTopic}/default/events/{DataSetWriterGroup}"
`

Memory snapshots after 5 min and  after 10 min already show a noticeable increase:

[image]

Corresponding Visual Studio `diagsession` file with memory snapshots (ZIP archive split into 6 parts—please remove the “.zip” extension from each downloaded part before extracting -  had to do this because of GitHub’s upload restrictions):

[Report20250526-1558.zip.001.zip](https://github.com/user-attachments/files/20442615/Report20250526-1558.zip.001.zip)
[Report20250526-1558.zip.002.zip](https://github.com/user-attachments/files/20442624/Report20250526-1558.zip.002.zip)
[Report20250526-1558.zip.003.zip](https://github.com/user-attachments/files/20442628/Report20250526-1558.zip.003.zip)
[Report20250526-1558.zip.004.zip](https://github.com/user-attachments/files/20442631/Report20250526-1558.zip.004.zip)
[Report20250526-1558.zip.005.zip](https://github.com/user-attachments/files/20442633/Report20250526-1558.zip.005.zip)
[Report20250526-1558.zip.006.zip](https://github.com/user-attachments/files/20442635/Report20250526-1558.zip.006.zip)

Please let me know if any additional details or alternative test setups would help.

quality-leftovers, 2025-05-26:
On first look: Might be missing Unregister, linking the ApplicationMessages to NetworkMessageSink CancellationTokenSource?

https://github.com/microsoft/project-furly/blob/14bb95d235f2e68d43d312face99bec9b98aed35/src/Furly.Extensions.Mqtt/src/Clients/MqttSession.cs#L231

Probably affecting all deployments using MQTT after porting to MQTT NET 5?

marcschier, 2025-06-03:
Thanks for the detailed analysis.  I am currently looking at other issues, but will prioritize this highest for next release.

marcschier, 2025-07-29:
In 2.9.15 I am testing @dimtix command line but with -t Null and see no memory increases.  I have fixed to dispose all cancellation token sources in mqtt, so if it is mqtt, we should also look better.  However, initial report was with IoT Hub, which I will also test next.

Haeavar, 2025-09-10:
After some more investigation it seems, that the memory leak is related to the OPC UA server restart. Our setup is an OPC UA aggregation server with multiple underlying OPC UA server which are restarted multiple times per day.
In a local debug profiler session I simulated the same. After each OPC UA server restart the memory consumption increased with a lot of dead objects:

[image]

For you information: This issue is **not in** version 2.9.11 but started with version 2.9.12. Maybe this helps to identify the root cause.

If you need more information, please let me now.

Haeavar, 2025-10-01:
@marcschier 
It seems, that the memory leak comes from the `InitializeMetrics()` within the `Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient`. If I comment the line `InitializeMetrics();` in the constructor  the memory consumption stays pretty much the same.

I'm wondering, why the `_meter` member is never disposed in the OpcUaClient? And why is the `ServiceCallContext.Dispose()` method does not dispose the `_client`?