Azure/Industrial-IoT #2366: Make Furly Extensions trimmer friendly

quality-leftovers, 2024-11-21:
**Is your feature request related to a problem? Please describe.**
To interact with the publisher API we are using the publisher models and the Furly.Extensions library so we don't have to maintain the models ourselves (mostly because Furly supports Variant)

In the future we'd like to be able to trim our clients to reduce memory consumption and application sizes, this is mostly blocked by our use of Furly.Extensions and the Variant type not being trim friendly.

**Describe the solution you'd like**
Json serialization / deserialization of publisher models should be trimmable. This would allow both end users and possibly the opc-publisher itself to compile .NET application with (partial) trimming reducing memory footprint and therefore hardware requirements / cost.

**Describe alternatives you've considered**
n.a.

**Additional context**
n.a.

marcschier, 2024-11-22:
Changed the title to what I would like to do.  Problem at this point is more timeline.

quality-leftovers, 2024-11-28:
Thanks for considering it.

New title is much better since Publisher has the whole OPCFouncation ComplexTypes dynamic type/assembly generation and will probably only be partially trimmable for a long time anyway.

closed 2026-09-05 by marcschier