Azure/Industrial-IoT #2370: Deprecate and remove cloud REST companion application

marcschier, 2025-01-17:
We are considering removing the REST companion web app and only ship the OPC Publisher IoT Edge module going forward. 

Is this a concern for anyone using OPC Publisher?

quality-leftovers, 2025-02-06:
At least for us it is not a concern. Albeit it is always nice to have a sample.

marcschier, 2025-06-13:
@NoTuxNoBux, the plan proposed here would be to stop releasing the "cloud webapp" container, which would include the Twin API you are referencing.  But this companion webapp just wraps the device twin and device methods behind a REST interface (with retries etc.) so apps in the cloud do not need to call any IoT Hub device methods directly.  So that you are not seeing the same behavior is surprising and would need to be looked at.  You could use the code in the web app in your Azure function, would that work?  

You could also use 2.9.14 of the container on MCR, but there would not be any more security fixes published, nor would we run end to end tests with 2.9.15+ of OPC Publisher then.  

Anyway, we could slice the webapp to just what you need and make it a sample, if that is ok for you, a list of API you are using would be great.

The core issue is that we just do not have cycles to update the deployment scripts and e2e devops pipelines to comply with security requirements for hosting on Microsoft build system. we are forced to turn off our builds/tests for this component at some point or muster time to stay in compliance. 

I want to say this too, so it is not misconstrued: The OPC Publisher is not affected by this at all, we did all the work to be compliant internally already.  It can still be controlled via Device Methods from IoT Hub, the API will not change, and no support timelines/conditions are affected, it remains fully integrated there and no changes are planned there.  The OPC Publisher module will also retain all its capabilities, nothing will be removed which includes its HTTPS API endpoint if you enable it.  

There was very limited feedback here until yesterday, nor did we ever get a support ticket on this component, which indicates that usage is very light.  It would be great if we can come up with a plan quickly that does not break anyone.

NoTuxNoBux, 2025-06-13:
Hi @marcschier Thanks for clarifying, in that case, if I understood correctly, we are indeed not affected by this change and don't mind it going away 🙂.

What we are doing right now is indeed calling HTTPS/REST API methods on the OPC Publisher directly on the IP address of the machine hosting it, [using ApiKey authentication](https://azure.github.io/Industrial-IoT/opc-publisher/transports.html#http). From what I understand now the cloud webapp is a separate application that we thus aren't using (we just deploy the OPC Publisher itself).

closed 2025-06-14 by marcschier