Azure/Industrial-IoT #2378: Opc Write take longer time

vity-topsoe, 2025-03-03:
OpcPublisher : 2.9.11

We are using https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/definitions.md#valuewriterequestmodelrequestenvelope

Request PAyload
`  var writeModeDataRequest = new
  {
      Connection = new
      {
          Endpoint = new
          {
              Url = opcPublisherConfig.EndpointUrl
          },
          User = RestAPIHelper.GetUserSchemaBasedOnAuthenticationType(opcPublisherConfig)
      },
      Request = new
      {
          nodeId = opcPublisherWrite.NodeId,
          value = opcPublisherWrite.Value,
          dataType=opcPublisherWrite.DataType
      }
  };

 var opcPublisherWriteRequest = System.Text.Json.JsonSerializer.Serialize(writeModeDataRequest);
  var content = new StringContent(opcPublisherWriteRequest, Encoding.UTF8, "application/json");
  var headers = new Dictionary<string, string>
  {
      { "Authorization", $"ApiKey  {apiKey}" }
  };
  return await _httpClientProvider.PostAsync($"{_opcPublisherWriteUrl}", content, headers, null).ConfigureAwait(false);

Container create options:
`

`{
  "Hostname": "publisher",
  "Cmd": [
    "publisher",
    "--pf=/appdata/publishednodes.json",
    "--lf=/appdata/publisher.log",
    "--aa",
    "--me=Json",
    "--mm=PubSub",
    "--fm=false",
    "--cfa=true",
    "--fd=false",
    "--bs=1",
    "--si=1",
    "--di=10",
    "--tm=True",
    "--ih=Mqtt",
    "--hb=3600",
    "--ot=30000",
    "--tp=/appdata/pki/trusted",
    "--rp=/appdata/pki/rejected",
    "--ap=/appdata/pki/own",
    "--ip=/appdata/pki/issuer"
  ],`

Write API  is taking 10 sec to write on Siemens OPC UA Server
[image]

UA Expertclient is able to write in miliseconds on Siemens OPC UA server.

Write API  is however taking1 sec to write on OPC Martrikon Simluation Server.

Do we need to to any Optimizations on OpcPublisher Module  so write api take less time.

Haeavar, 2025-03-03:
@vity-topsoe  You could try the connection option `  "options": "NoComplexTypeSystem"` to disable complex type system loading. As you see in your logs, this takes serval seconds....

vity-topsoe, 2025-03-04:
> [@vity-topsoe](https://github.com/vity-topsoe) You could try the connection option ` "options": "NoComplexTypeSystem"` to disable complex type system loading. As you see in your logs, this takes serval seconds....

@Haeavar Thank you it reduced the time little but is still high.

If we see logs below from 05:17:49 it takes 3 seconds accepting certificate etc .
It it took 5:17:49 to 5:17:57 can write use existing connection or any other way to optimize it ?
We need the response in milliseconds as it id doing with UAExpert.

[25-03-04 05:17:49.3376] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      => SpanId:894e1bba4df53c9d, TraceId:35b7c6f988b712ac8372086347b03c80, ParentId:0000000000000000 => ConnectionId:0HNANQH0PHMK4 => RequestPath:/v2/write RequestId:0HNANQH0PHMK4:00000001 => Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController.ValueWriteAsync (Azure.IIoT.OpcUa.Publisher.Module)
      Accepting untrusted peer certificate C3D11873AD342A7D0CCDEA92ACC32772736B70EB, 'CN=UC.Q01-K422/OPCUA-1-2, OU=Industrial Communication, O=Siemens, C=DE' due to AutoAccept(UntrustedCertificates) set!
[25-03-04 05:17:54.2009] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      => SpanId:894e1bba4df53c9d, TraceId:35b7c6f988b712ac8372086347b03c80, ParentId:0000000000000000 => ConnectionId:0HNANQH0PHMK4 => RequestPath:/v2/write RequestId:0HNANQH0PHMK4:00000001 => Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController.ValueWriteAsync (Azure.IIoT.OpcUa.Publisher.Module)
      Accepting untrusted peer certificate C3D11873AD342A7D0CCDEA92ACC32772736B70EB, 'CN=UC.Q01-K422/OPCUA-1-2, OU=Industrial Communication, O=Siemens, C=DE' due to AutoAccept(UntrustedCertificates) set!
[25-03-04 05:17:57.0125] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      => SpanId:894e1bba4df53c9d, TraceId:35b7c6f988b712ac8372086347b03c80, ParentId:0000000000000000 => ConnectionId:0HNANQH0PHMK4 => RequestPath:/v2/write RequestId:0HNANQH0PHMK4:00000001 => Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController.ValueWriteAsync (Azure.IIoT.OpcUa.Publisher.Module)
      opc.tcp://severip:4840/Leak02_EM03_QC_D77126A9 [state:Ready|refs:1]: Session opc.tcp://severip:4840/Leak02_EM03_QC_D77126A9 with opc.tcp://severip:4840/Leak02_EM03_QC changed from Connecting to Ready
[25-03-04 05:17:57.0125] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      => SpanId:894e1bba4df53c9d, TraceId:35b7c6f988b712ac8372086347b03c80, ParentId:0000000000000000 => ConnectionId:0HNANQH0PHMK4 => RequestPath:/v2/write RequestId:0HNANQH0PHMK4:00000001 => Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController.ValueWriteAsync (Azure.IIoT.OpcUa.Publisher.Module)
      Channel 2940527348 got new token 1 (03/04/2025 05:17:49).
[25-03-04 05:17:57.0126] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      => SpanId:894e1bba4df53c9d, TraceId:35b7c6f988b712ac8372086347b03c80, ParentId:0000000000000000 => ConnectionId:0HNANQH0PHMK4 => RequestPath:/v2/write RequestId:0HNANQH0PHMK4:00000001 => Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController.ValueWriteAsync (Azure.IIoT.OpcUa.Publisher.Module)
      Channel diagnostics for session ns=1;i=1694949637 updated.

marcschier, 2025-03-07:
Use --cl=<seconds> to let the client sessions linger in the publisher for the amount of seconds before they get closed.  Then any calls within the linger period will keep the session alive and re-use it.