Azure/Industrial-IoT #2389: Cannot build the code locally

xky0007, 2025-03-26:
1. Git clone the code and open with vs 2022. Got following errors:
`Unable to load the service index for source https://pkgs.dev.azure.com/msazure/One/_packaging/Industrial-IoT/nuget/v3/index.json.
  Response status code does not indicate success: 401 (Unauthorized).`

How to fix this? Thanks.

xky0007, 2025-03-26:
[image]

xky0007, 2025-03-26:
Seems my nuget config crashed. Fixed.

marcschier, 2025-04-17:
For Microsoft to ensure supply chain security, we must consume from an internal aggregator feed.  The nuget.config unfortunately has to specify it or our build policies fail.  The aggregator feed is not accessible to the community, but the simple fix is to delete the nuget.config in the root of the repo just like the GitHub Actions do.