Azure/Industrial-IoT #2390: Add support for Compact and Verbose Encoding

quality-leftovers, 2025-03-26:
**Is your feature request related to a problem? Please describe.**
OPC UA Specification introduced "new" encodings.

**Describe the solution you'd like**
Add support Compact and Verbose Encoding, including combincation of Verbose & Raw (Verbose without UaType and UaTypeId fields).

**Describe alternatives you've considered**
-

**Additional context**

From the spec:

The expectation is all encoders used for Publishers support all four encodings and all decoders for Subscribers support the ReversibleEncoding, CompactEncoding and VerboseEncodings.

The CompactEncoding and VerboseEncoding replace the ReversibleEncoding and NonReversibleEncoding. The differences are described in Annex H. The VerboseEncoding also supports the RawData mode defined in OPC 10000-14. In RawData mode, encoders shall omit the following fields:

UaType (see 5.4.2.17 and 5.4.2.18);

UaTypeId (see 5.4.2.16);

...

quality-leftovers, 2025-03-26:
**Format Samples**
I tried finding some samples for the encoding, but the samples in Part 14: PubSub looked a bit stange since they never had a `UaTypeId`. So I used the OPC Foundation unit tests to produce samples.

Given a structure like:
```
{
   string Foo = "bar_1";
   string Foor2 = "bar_2";
   int Foo3 = 0;  
}
```

Compact encoded Json:
```json
{"Foo":"bar_1","Foo2":"bar_2"}
```

Verbose encoded Json:
```json
{"UaTypeId":"s=test_dyn2_typeid","Foo":"bar_1","Foo2":"bar_2","Foo3":0}
```

Verbose + Raw encoded Json (?):
```json
{"Foo":"bar_1","Foo2":"bar_2","Foo3":0}
```

marcschier, 2025-03-26:
We will evaluate this for a future release.

closed 2026-09-05 by marcschier