Azure/Industrial-IoT #2416: OPC Publisher does not re-establish connection to OPC UA server, if the OPC UA server is restarted.

TwoPatient, 2025-08-21:
**Describe the bug**
OPC Publisher does not re-establish connection to OPC UA server, if the OPC UA server is restarted. Using a 3rd party OPC Client reconnects without issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup OPC Publisher 2.9.14 running on IoT Edge EFLOW 1.5 connected to a separate server running Node Red OPC UA Server.
2. OPC Publisher setup to read NodeRed OPC UA.
3. OPC Publisher receives data changes.
4. Restart server running Node Red.
5. OPC Publisher does not receive data changes post restarting OPC UA server.**Expected behavior**
OPC Publisher should re-establish connection to OPC UA server and receive data changes

**Desktop (please complete the following information):**
 - EFLOW IoT Edge 1.5
 - OPC Publisher 2.9.14
 - NodeRed 4.0.9

MartonBoda, 2025-08-26:
Hi, attached opc publisher trace logs.
[publisherlogs_trace_1.log](https://github.com/user-attachments/files/21984605/publisherlogs_trace_1.log)

It shows initially everything was fine,
then the OPC server was restarted,
the OPC Publisher reconnected, but the nodes stayed bad.

The OPC Publisher starts collecting data after manually restarting it.

MartonBoda, 2025-08-27:
Thank you, I've shared the publishednodes.json and a log with Information level and --ln flag.

[publishednodes_20250827.json](https://github.com/user-attachments/files/22004632/publishednodes_20250827.json)
[publisherlogs-202508270930.log](https://github.com/user-attachments/files/22004619/publisherlogs-202508270930.log)

MartonBoda, 2025-08-28:
I've tried it, the preview version has the same behaviour.
[publisherlogs-2-9-15-preview4-20250828-1.log](https://github.com/user-attachments/files/22033323/publisherlogs-2-9-15-preview4-20250828-1.log)

marcschier, 2025-09-17:
As we found, the node.red server loads nodes after starting, but also after publisher tries to establish the subscription.  Therefore the publisher needs to be configured to retry.  Retry was broken and has now been fixed in #2420.  This will be released in 2.9.15.